zephyrproject-rtos/zephyr #9772: Test application hangs without any console output on x86/ARC based boards @arduino_101:arc

pswarnak, 2018-09-03:
All application hang on ARC core without any console log or error message.

**Arch:** arc
**Board:** arduino_101_sss, quark_se_c1000_ss_devboard
**Zephyr commit:** c8402bc3cee3d53493876fd19166bda273280b86
**Toolchain:**  ZEPHYR_TOOLCHAIN_VARIANT=zephyr
**Additional Info:** CONFIG_ARC_INIT=y was removed by @inakypg in commit b073b296889115c175230f98537095c7fb66bbe9.

**Error:**
No console log available at all.

**Steps to reproduce:**
```
**For running application on the ARC core:**
cd zephyr/samples/hello_world/
rm -rf build && mkdir build && cd build
cmake -D BOARD=arduino_101_sss ../
make BOARD=arduino_101_sss flash

**Additionally, flash the boot stub to enable messages from the ARC processor:**
cd zephyr/tests/booting/stub 
cmake -D BOARD=quark_se_c1000_devboard  ../
make BOARD=arduino_101  flash 

Check the console log.
```

**Sample Expected output:**
```
Hello World! arc
```

nashif, 2018-09-03:
this is an issue in TCF.

inakypg, 2018-09-04:
@pswarnak  which test applciation?

pswarnak, 2018-09-04:
@inakypg:
I tried multiple applications, none worked.
You can try hello_world sample test as mentioned in "Steps to reproduce".
This fails only on local setup and flash, not on TCF.

inakypg, 2018-09-04:
blink blink
so then, you are missing the *board* specific step of adding whichever configuration the stub needs for it to work? I think this is a case of PEBKAC.

The stub's CONFIG_ARC_INIT was removed because it was Quark specific -- the entity that's building the stub needs to decide what config options it has to set for its intended deployment.

In this case, *you*, as you are running it manually, need to set CONFIG_ARC_INIT when loading the stub on the Quark's x86 core.

PR #9807 (2018-09-05): boards: arduino_101: remove mention of boot stub
The stub is used for testing only and is not compatible anymore with the
usage documented here.

Fixed #9772
commit d3bdc851 (2018-09-05): boards: arduino_101: remove mention of boot stub

The stub is used for testing only and is not compatible anymore with the
usage documented here.

Fixed #9772
closed 2018-09-04 by inakypg