zephyrproject-rtos/zephyr #9777: tests/kernel/mem_pool/mem_pool_api:mpool_alloc_timeout crashes on qemu_riscv32 with boot delay

spoorthik, 2018-09-04:
This test case fails only when boot delay is introduced.

Execution log:
```
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.13.0-rc2-15-gb68cf99 (delayed boot 1000ms) *****
Running test suite mpool_api
===================================================================
starting test - test_mpool_alloc_free_thread
PASS - test_mpool_alloc_free_thread
===================================================================
starting test - test_mpool_alloc_free_isr
PASS - test_mpool_alloc_free_isr
===================================================================
starting test - test_mpool_kdefine_extern
PASS - test_mpool_kdefine_extern
===================================================================
starting test - test_mpool_alloc_size
PASS - test_mpool_alloc_size
===================================================================
starting test - test_mpool_alloc_timeout
Exception cause Illegal instruction (2)
Current thread ID = 0x80005024
Faulting instruction address = 0x8
  ra: 0x0  gp: 0x0  tp: 0x0  t0: 0x0
  t1: 0x0  t2: 0x0  t3: 0x0  t4: 0x0
  t5: 0x0  t6: 0x0  a0: 0x800025dc  a1: 0x80005110
  a2: 0x0  a3: 0x0  a4: 0x0  a5: 0x80001610
  a6: 0x80  a7: 0x20
Fatal fault in essential thread! Spinning...

```

Latest commit on which failure is seen:  b68cf999d663a8ddd75432f0293ac4c1d341af4f

Steps to reproduce:
```
cd zephyr.git/tests/kernel/mem_pool/mem_pool_api
Append CONFIG_BOOT_DELAY=1000 in prj.conf file
mkdir build && cd build
cmake -DBOARD=qemu_riscv32 ..
make run
```

galak, 2018-12-07:
@nategraff-sifive can you look at this, I was able to reproduce it based on the instructions in the issue.

nategraff-sifive, 2018-12-07:
I'll take a look. Also, I can reproduce this failure on latest master (3651e03eebbba5a8d281348d145ef8e03d8b65db) with no boot delay:

```
***** Booting Zephyr OS zephyr-v1.13.0-2466-g3651e03 *****
Running test suite mpool_api
===================================================================
starting test - test_mpool_alloc_free_thread
PASS - test_mpool_alloc_free_thread
===================================================================
starting test - test_mpool_alloc_free_isr
PASS - test_mpool_alloc_free_isr
===================================================================
starting test - test_mpool_kdefine_extern
PASS - test_mpool_kdefine_extern
===================================================================
starting test - test_mpool_alloc_size
PASS - test_mpool_alloc_size
===================================================================
starting test - test_mpool_alloc_timeout
Exception cause Illegal instruction (2)
Current thread ID = 0x80000208
Faulting instruction address = 0x8
  ra: 0x0  gp: 0x0  tp: 0x0  t0: 0x0
  t1: 0x0  t2: 0x0  t3: 0x0  t4: 0x0
  t5: 0x0  t6: 0x0  a0: 0x204030e4  a1: 0x800002e8
  a2: 0x0  a3: 0x0  a4: 0x0  a5: 0x20401500
  a6: 0x80  a7: 0x20
Fatal fault in essential thread! Spinning...
```

nategraff-sifive, 2018-12-07:
It looks like the failure is actually related to `test_mpool_alloc_free_isr`.

If you modify `test_main()` to just run `test_mpool_aloc_free_isr`, you get 

```
***** Booting Zephyr OS zephyr-v1.13.0-2466-g3651e03 *****
Running test suite mpool_api
===================================================================
starting test - test_mpool_alloc_free_isr
PASS - test_mpool_alloc_free_isr
===================================================================
Test suite mpool_api succeeded
===================================================================
PROJECT EXECUTION SUCCESSFUL
Exception cause Illegal instruction (2)
Current thread ID = 0x800001a8
Faulting instruction address = 0x8
  ra: 0x0  gp: 0x0  tp: 0x0  t0: 0x0
  t1: 0x0  t2: 0x0  t3: 0x0  t4: 0x0
  t5: 0x0  t6: 0x0  a0: 0x20402c60  a1: 0x80000288
  a2: 0x0  a3: 0x0  a4: 0x0  a5: 0x2040107c
  a6: 0x80  a7: 0x20
Fatal fault in essential thread! Spinning...
```

nategraff-sifive, 2018-12-07:
@galak I can fix this by modifying prj.conf to increase the ISR stack size from 512 to 1024 bytes. Is that an acceptable solution?

andrewboie, 2018-12-07:
Looking at the code, it might be simplest to just define the array of blocks at the top of the function as static, so they don't go on the call stack.

PR #11962 (2018-12-08): tests: mem_pool_api: reduce stack usage
Some arches were getting a stack overflow in ISR context.

Fixes: #9777
commit 5f793cc2 (2018-12-08): tests: mem_pool_api: reduce stack usage

Some arches were getting a stack overflow in ISR context.

Fixes: #9777
closed 2018-12-08 by nashif