zephyrproject-rtos/zephyr #9778: Implement zephyr specific SEGGER_RTT_LOCK and SEGGER_RTT_UNLOCK macros

nordic-krch, 2018-09-04:
In Segger_RTT_Conf.h there are macros for locking access to RTT:
https://github.com/zephyrproject-rtos/zephyr/blob/b68cf999d663a8ddd75432f0293ac4c1d341af4f/ext/debug/segger/rtt/SEGGER_RTT_Conf.h#L142

They are used for example in SEGGER_RTT_Write() but are omitted when SEGGER_RTT_WriteNoLock() is used. Currently console is using SEGGER_RTT_WriteNoLock() and provides own lock (irq_lock). #9365 adds RTT logger backend which assess RTT as well. Currently it must use same approach (irq_lock) to avoid collision and this approach will fail if longer buffers are used (critical interrupts locked).

Locking using mutex should be implemented in SEGGER_RTT_LOCK/UNLOCK.

pkral78, 2018-10-17:
#10659 make RTT backend independent on rtt console, when console is also used then rtt backend may by configured to differend up-buffer separating terminal from logger, no locking needed.

In case when both console and logger access terminal buffer (which is not forbidden) we of course need mutex.

pkral78, 2018-10-17:
Sharing now implemented in #10659 by commit https://github.com/zephyrproject-rtos/zephyr/pull/10659/commits/e1f3fc87198d43a91fa6bd42c9599ca729641570

nashif, 2021-05-28:
This is already addressed

closed 2021-05-28 by nashif