zephyrproject-rtos/zephyr #9783: LwM2M: Cannot create object instance without specifying instance number

therealprof, 2018-09-04:
The LwM2M TS says:

> When there is no reference to Object Instance in the TLV/CBOR/JSON payload of the "Create" command, the LwM2M Client MUST assigns the ID of the created Object Instance. 

However the Zephyr implementation does not have logic to pick a suitable object instance in `do_write_op_tlv()` and thus simply does nothing. And to add insult to the injury it'll happily lie to the server by sending back a "2.01 Created" response.

nashif, 2018-09-04:
@mike-scott can you please take a look?

mike-scott, 2018-09-04:
@nashif Yep, I'll check it out

mike-scott, 2019-02-27:
@therealprof I'm assuming this is still an issue... But I want to confirm:
When I look at:
https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/net/lib/lwm2m/lwm2m_rw_oma_tlv.c#L885

I see the "escape" clause where if there aren't any TLV items, we basically abort but return 0.  Is that what you're hitting?

mike-scott, 2019-03-05:
@nashif Let's set the priority of this to low.  Most servers won't actually let you submit this kind of request to the client.

We will fix it, but I need to come up with a decent way of testing the fix once I do (I don't have access to any servers which allow me to do this).

closed 2020-10-06 by github-actions[bot]