zephyrproject-rtos/zephyr #9831: 1.14 Merge Order

galak, 2018-09-06:
This Issue is to track which changes should try and get merged in which order early on so we can reduce churn & conflicts:

Topic | Merge Status | impact to code base | PR
----- | -------------- | ------------------- | ---
Move SoC code to top-level | ready | impacts new SoC supports in arch/*/soc | #9776 
DTS code generation | not ready | 
CMake library generation changes | 
Moving to new Logging subsystem |   | numerous PRs | Merge PR #9450 first, [logging PRs](https://github.com/zephyrproject-rtos/zephyr/pulls?q=is%3Apr+is%3Aopen+soc+label%3A%22area%3A+Logging%22)
Moving to new shell subsystem | 
Multi-Repo, west tool |

mike-scott, 2018-09-06:
Where does the move to socket API support in the networking subsystems land in this list?

nashif, 2018-09-06:
nice table :)

nashif, 2018-09-06:
> Where does the move to socket API support in the networking subsystems land in this list?

which PR is that?

mike-scott, 2018-09-06:
@nashif this is the primary tracking issue for socket API support: https://github.com/zephyrproject-rtos/zephyr/issues/7591  Hopefully, before the weekend, I plan to submit an RFC/Proof-of-concept PR changing CoAP and LwM2M so that they support *both* sockets and net-app (based on Kconfig).

carlescufi, 2018-09-06:
@galak added some additional info, thanks for creating this.

nashif, 2018-09-06:
@mike-scott 
> @nashif this is the primary tracking issue for socket API support: #7591 Hopefully, before the weekend, I plan to submit an RFC/Proof-of-concept PR changing CoAP and LwM2M so that they support both sockets and net-app (based on Kconfig).

This is more or less for everything that is either complete or semi-complete and  intrusive in nature, i.e. touches multiple components. 
We are trying to avoid lots of churn by prioritising review/merge and focusing on the high priority PRs we want to get in for 1.14.

Everything else continues as usual.

pfalcon, 2018-09-06:
> and intrusive in nature

That's how I understand this list too. For sockets, stuff-before-sockets remains there for the foreseeable future, so socket tasks aren't for this list. (Unlike e.g. logging, which, if we decided to switch too, needs to be done, or it's really a pain to deal with 2 systems).

@mike-scott ^^

mike-scott, 2018-09-06:
ok

mbolivar, 2018-09-06:
I'm confused about the relationship between the various pull requests regarding code generation:

- #8561 (named in the above table): Generated driver instantiation prototype
- #9406: Generate DTS 'compatible' based compilation flags
- #6762: scripts: add Zephyr inline code generation with Python
- #9163: drivers: use codegen for device instantiation
- #5799: drivers: pinctrl device interface (Linux-like) and STM32F0 driver

I'm also confused about what the problems that are trying to be solved are, and which of these PRs are dependencies versus competing implementations.

Variously, I see attempts to:

- add a Python --> C code generation library (after abandoning a "CHAOS" C header only attempt)
- standardize pin muxing using code generation
- convert DTS to JSON for various purposes, including as an input to the code generation library
- generate device instantiation code from JSON + code generation
- control device driver build system flags based on compatible flags in DTS
- probably more that I've missed

There isn't an umbrella issue on GitHub named in the table here that describes the problem, solution space, etc.

Can someone point me in the right direction? There appears to be quite a lot of vigorous discussion in the PRs and it's overwhelming.

If this is going to be an intrusive change, it would be great to have a single place with a clear summary of the problem, proposed solutions, etc., especially if at least the problem statement has buy-in from all the parties involved.

erwango, 2018-09-07:
@mbolivar , agreed this global pictures miss clarity. Though, a umbrella issue has been created: #8499,
and this should be the reference.
Let's discuss in this one in order not to pollute current issue.

mbolivar, 2018-09-07:
> @mbolivar , agreed this global pictures miss clarity. Though, a umbrella issue has been created: #8499,
> and this should be the reference.
> Let's discuss in this one in order not to pollute current issue.

Thank you @erwango for the clarification! I will study the issue in #8499.

@galak could you please add the issues related to each PR to your table if it's not too much trouble? I imagine this table will be read by many people since it's the comprehensive list of highly intrusive changes for v1.14, so it should be worth the effort.

Olivier-ProGlove, 2018-09-11:
Being the author of numerous logging pull-requests, I am also waiting for @nordic-krch's pull-request: https://github.com/zephyrproject-rtos/zephyr/issues/9761 to be reviewed to rebase and refactor my pull-requests on `master`.

nashif, 2018-09-11:
meeting minutes:

- soc tree move to top level: missing docs, ready to go as first PR, no major PRs in progress that would block it
- Logging: setup a meeting with developers and users and come up with a plan
- DTS: 
  - review EDTS and merge (this is important and perquisite for many changes, so it needs to be dealt with highest priority.

nashif, 2018-09-12:
one other big and intrusive change:
- libc and type changes.

closed 2019-04-12 by nashif