zephyrproject-rtos/zephyr #9843: tests/kernel/sched/deadline fails on NRF5x boards

spoorthik, 2018-09-07:
Commit: 02aa9800424453f4442e279048c84b9e4266063e

Execution log:

```
 ***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS 1.13.0-rc2 (delayed boot 1000ms) *****
Running test suite suite_deadline
===================================================================
starting test - test_deadline

    Assertion failed at /zephyr.git/tests/kernel/sched/deadline/src/main.c:70: test_deadline: (n_exec == 0 is false)
threads ran too soon
FAIL - test_deadline
===================================================================
===================================================================

PROJECT EXECUTION FAILED
```

Platforms tested: nrf52840_pca10056:arm, nrf52_pca10040:arm, nrf51_pca10028:arm

andyross, 2018-09-07:
Always NRFx...  It's true that I underconstrained the scheduler setup -- maybe the main thread is already at the lowest priority on those boards?

nashif, 2018-09-07:
CONFIG_BT=n and it works...
Something in the BT stack interferes with the test, suggest to add CONFIG_BT=n to the prj.conf

andyross, 2018-09-07:
That's a good catch.  Yeah, there are threads and priority selection done in the context of the BT and network layers, if it pushes the main thread to the bottom that could do it.  If it doesn't, though, nothing should really "interfere" with the scheduling here.  Agree with that for an immediate fix, but I should dig a little before closing this.

andyross, 2018-09-07:
Suggested fix submitted, but please don't close until I can figure out why this is happening (strictly: it's not a problem with the EDF feature, the detected case is that the threads are running before the test's main entry point has blocked, which shouldn't happen).   A quick look at the generated .config doesn't show anything obviously wrong.

pizi-nordic, 2018-09-12:
@carlescufi, @cvinayak: The https://github.com/zephyrproject-rtos/zephyr/pull/9745 might be source of this problem.

carlescufi, 2018-09-12:
@nashif @andyross as per @pizi-nordic suggestion, this might be caused by the fix we posted for the 1.13 release which fixes some tests, but makes others fail. The only clean solution is to follow-up on #9904.

andyross, 2018-11-15:
Should be fixed now.  Can't reproduce on nrf52840_pca10056 post-timer-rewrite.

andyross, 2019-03-01:
Hasn't reappeared.  Must be fixed.  Closing.

PR #9855 (2018-09-08): tests/kernel/sched/deadline: Disable CONFIG_BT
The BT threads are interfering with the main thread priority selection
and the test fails semi-spuriously on NRF5x boards with "threads ran
too soon" (i.e. not an EDF failure per se, but the fact that the new
threads at K_LOWEST_APPLICATION_PRIO are running instead of the test
thread).  This is a reasonable workaround for testing the
SCHED_DEADLINE ordering behavior, though.

Fixes #9843
commit a7e5d2f0 (2018-09-08): tests/kernel/sched/deadline: Disable CONFIG_BT

The BT threads are interfering with the main thread priority selection
and the test fails semi-spuriously on NRF5x boards with "threads ran
too soon" (i.e. not an EDF failure per se, but the fact that the new
threads at K_LOWEST_APPLICATION_PRIO are running instead of the test
thread).  This is a reasonable workaround for testing the
SCHED_DEADLINE ordering behavior, though.

Fixes #9843
closed 2019-03-01 by andyross