zephyrproject-rtos/zephyr #9862: tests/drivers/build_all#test_build_sensors_a_m @ quark_se_c1000_devboard:x86 BUILD failed

pswarnak, 2018-09-09:
The test **tests/drivers/build_all** results into build failure on quark_se_c1000_devboard for test_build_sensors_a_m and test_build_sensor_triggers, yaml configs.

**Arch:** quark_se_c1000_devboard
**Board:** x86
**Zephyr version:**  commit 8250b3e544131392bff2c794b85b7b93d9388587.
**Toolchain:**  ZEPHYR_TOOLCHAIN_VARIANT=zephyr

**Build Log:**
```
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.5.4", minimum required is "3.4")
-- Selected BOARD quark_se_c1000_devboard
Zephyr version: 1.13.0
Parsing Kconfig tree in /home/pswarnak/workspace/1.13_execution/zephyr/Kconfig
Loading /home/pswarnak/workspace/1.13_execution/zephyr/boards/x86/quark_se_c1000_devboard/quark_se_c1000_devboard_defconfig as base
Merging /home/pswarnak/workspace/1.13_execution/zephyr/tests/drivers/build_all/prj.conf
-- Generating zephyr/include/generated/generated_dts_board.h
quark_se_c1000_devboard.dts_compiled: Warning (unit_address_vs_reg): Node /soc/virtualcom@0 has a unit name, but no reg property
extract_dts_includes.py: 'id' property missing in 'Intel I/O Advanced Programmable Interrupt Controller' binding. Using '<unknown id>'.
-- Cache files will be written to: /home/pswarnak/.cache/zephyr
-- The C compiler identification is GNU 6.2.0
-- The CXX compiler identification is GNU 6.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/i586-zephyr-elfiamcu/i586-zephyr-elfiamcu-gcc
-- Performing Test toolchain_is_ok
-- Performing Test toolchain_is_ok - Success
CMake Error at ../../../CMakeLists.txt:527 (message):
  The Zephyr library 'drivers__adc' was created without source files.  Empty
  (non-imported) libraries are not supported.  Either make sure that the
  library has the sources it should have, or make sure it is not created when
  it has no source files.

-- Configuring incomplete, errors occurred!
See also "/home/pswarnak/workspace/1.13_execution/zephyr/tests/drivers/build_all/build/CMakeFiles/CMakeOutput.log".
See also "/home/pswarnak/workspace/1.13_execution/zephyr/tests/drivers/build_all/build/CMakeFiles/CMakeError.log".
```

**Steps to reproduce:**
```
cd zephyr/tests/lib/mem_alloc/
cp sensors_a_m.conf prj.conf
OR
cp sensors_trigger.conf prj.conf
rm -rf build && mkdir build && cd build
cmake -D BOARD=quark_se_c1000_devboard -GNinja ../

Check the build log.
```

nashif, 2018-09-09:
you are skipping the filter here, this should not build on platforms without adc. the test with this config depends on the board having ADC support.

closed 2018-09-09 by nashif

pswarnak, 2018-09-09:
Sorry missed it, thanks for info.