zephyrproject-rtos/zephyr #9873: [EXT. FLASH] Driver for the MX25Rxx

GiulianoFranchetto, 2018-09-10:
Hi,

The Nordic nRF52840 PDK has an external flash built-in, connected to the nRF52840 by QSPI.

The driver is available using the nordic SDK, but it's not in Zephyr yet. 

Has anybody already ported this driver on Zephyr, or @carlescufi does the Nordic team plan to do so?

We do not need a XIP-enabled driver, we just want to read-write on the flash.
 
Regards

GiulianoFranchetto, 2018-09-17:
Hi,

any update on this issue?

carlescufi, 2018-09-17:
Copying @kl-cruz who has more details on this flash.

GiulianoFranchetto, 2018-09-27:
Hi,

The QSPI part is not really what we want for now, as we don't have code in the external flash.

Did Nordic developped a driver for the  MX25Rxx when creating the nrf52 PDK?

Could it be easily ported using the Zephyr API?

Regards

kl-cruz, 2018-09-28:
Hi,
I think QSPI could fit to your requirements. You do not have to use XiP. Just initialize QSPI and read/write/erase data from external memory. There is example application in Nordic's SDK (for example 15.2.0) based on QSPI. I think it is good starting point  to write shim layer.
In my opinion using QSPI to drive MX25xxx memories is easier than preparing SPI driver

nvlsianpu, 2018-09-28:
@kl-cruz is this somehow related? https://github.com/zephyrproject-rtos/zephyr/pull/7561

findlayfeng, 2018-11-02:
@kl-cruz 
Reading sfdp using QSPI first is defined in the sfdp documentation.
And this attempt is only done during initialization, and the result can be returned immediately without waiting. It has little effect on the initialization time.
And for devices that only support spi to read sfdp, this is a more expensive attempt, but it will be faster for devices that support qspi to read sfdp.

nvlsianpu, 2020-02-13:
QSPI based driver Introduced as Nordic QSPI nor driver (usage examples tests/subsys/fs/littlefs on nrf52840_pca10056). 
https://github.com/zephyrproject-rtos/zephyr/pull/21535

Also possible to use spi nor driver.

closed 2020-02-13 by nvlsianpu