zephyrproject-rtos/zephyr #9879: Portability: Arithmetic on void pointers

mped-oticon, 2018-09-10:
Hi,

One use of GCC extensions we are patching-away is the use of arithmetic on void pointers [1].
PR will follow soon, but this issue is raised to also discourage future use of this extension.

Arithmetics on void pointers are used in bluetooth/gatt.h and mempool.

We patch this as our clang-based compiler only supports standard C99, C11.
Note that our compiler is provided by one of the major EDA vendors.  While this is a trivial issue, it affects portabillity and patching incurs legal issues with respect to Certification, FDA, etc.

Best regards, 
Mark

[1] https://gcc.gnu.org/onlinedocs/gcc/Pointer-Arith.html

jhedberg, 2018-09-11:
Could you clarify where in gatt.h you see this happening?

mped-oticon, 2018-09-11:
@jhedberg Sorry, I meant gatt.c: 
```
ssize_t bt_gatt_attr_read( .... u16_t offset, const void *value,  ....) {
  memcpy(buf, value + offset, len);
```
I will soon submit a PR.

mped-oticon, 2018-09-28:
Fixed as of PR #9917. Also added warning flag to prevent this from occurring in future.

closed 2018-09-28 by mped-oticon