zephyrproject-rtos/zephyr #9882: MISRA-C - Do not use reserved names

ceolin, 2018-09-10:
The C standard reserves all identifiers that begin with an underscore for implementation details. Zephyr is using underscore for identifiers and macros all over the project resulting in many violations. Bellow a suggestion to handle this:
   - If a function is static, it is renamed so that the leading underscore is removed
   - If a function isn't static, a rule based on its name is used to rename the function based on its name:
      - _arch_ -> z_arch_
      - _sys_ -> z_sys_
      - _k_ -> z_
      - _impl_ -> z_sys_impl_
      - _handler_ -> z_sys_handl_
      - Otherwise, z_
      - Shortcuts such as "_current", "_ready_q", "_timeout_q" and "_threads" can be changed to their respective values.

According to the C Standard, 7.1.3 [ISO/IEC 9899:2011]

  - All identifiers that begin with an underscore and either an uppercase letter or another underscore are always reserved for any use.
  - All identifiers that begin with an underscore are always reserved for use as identifiers with file scope in both the ordinary and tag name spaces.
  - Each macro name in any of the following subclauses (including the future library directions) is reserved for use as specified if any of its associated headers is included, unless explicitly stated otherwise.
  - All identifiers with external linkage (including future library directions) and errno are always reserved for use as identifiers with external linkage.

Part of #9552

andrewboie, 2018-12-14:
```
impl -> z_sys_impl_
handler -> z_sys_handl_
```

What's the semantics of the _sys_ in the middle?
Can we just do:

```
impl -> z_impl_
handler -> z_handl_
```

andyross, 2019-02-13:
I missed the discussion earlier and was pointed here after the @pfl had already submitted.

Regarding conventions, not that both _impl and _handler are only ever applied to syscalls, which are symbols in the public headers that **already** have (or should have, prior to this patch) well-formed namespaced names like "k_yield()" or whatever.  I'd argue we don't need the "z" at all, and it's perfectly acceptable to do a mapping like `k_yield` -> `impl_k_yield`  (or "k_impl_yield" or "kimpl_yield", ...)

Would also generally prefer to save bytes by not stuffing underscores into the middle of a prefix, so e.g. "zarch_" and "zsys_" make more sense to me.

pfl, 2019-02-13:
The original renaming list was written down by @ceolin so I'd guess it can be discussed and the scripts updated accordingly.

pfl, 2019-02-15:
Shall we go for

```
`_impl_` -> `z_impl_`
`_handler_` -> `z_handl_`
```
here?

andrewboie, 2019-02-15:
> Shall we go for
> 
> ```
> `_impl_` -> `z_impl_`
> `_handler_` -> `z_handl_`
> ```
> here?

Works for me

ceolin, 2019-02-15:
+1

 it seems better go with
```
_impl_ -> z_impl
_handler_ -> z_handl_
```

closed 2021-05-28 by nashif