zephyrproject-rtos/zephyr #9883: DTS processing generates unit_address_vs_reg warning on entries with 'reg'

dleach02, 2018-09-10:
Current DTS processing generates a warning on nodes that specify an unit-address via the '@<n>' but don't have a 'reg' property. 

According to the dts specification (section 2.2.1): 

> If the node has no reg property, the @unit-address must be omitted and the node-name along differentiates the node from other nodes at the same level in the tree.

One way to get rid of the warning is to suppress the warnings on the command line. The other way is to follow the specification and get rid of the @<n> and use something else that ensures node uniqueness. The problem with that is that our python scripts then creates improper header defines for the '<>_LABEL' for these nodes that don't have the '@'.

I've narrowed this down to how scripts/dts/extract/globals.py:get_node_label() builds the '<>_LABEL' define. It tests for the '@' and does a split on it to extract the <n> value after the '@'. This <n> would be unique among the set of nodes. But to if you want to avoid the warning, you need to remove the '@' sign and to still have uniqueness you would do something like led1 or led_1 instead of led@1 in the dts file. But get_node_label() then creates a string on the node_address string which includes '/' path information. A dts entry with 'led1' will result in a LABEL generation of: 

```
#define GPIO_LEDS/LEDS/LED1_LABEL		"User LD2"
```

vs 'led@0' dts entry generating:

```
#define GPIO_LEDS_0_LABEL		"User LD1"
```

A fix to this is to just extract the unique part of the node_address with:

`def_label += '_' + convert_string_to_label(node_address.split('/')[-1])
`

which is similar to a change performed on the '@<n>' format a while back. It results in a <>_LABEL define of:

`#define GPIO_LEDS_LED1_LABEL			"User LD2"
`

I'm unsure though if there are cascading impacts to changing the _LABEL name like this but this seems like a reasonable fix.

I'll prepare a PR for review.

erwango, 2018-09-17:
This point has been discussed in #7013 and current proposed solution for this issue is: https://github.com/zephyrproject-rtos/zephyr/issues/7564.

erwango, 2018-09-17:
If we agree on proposed solution, I propose to raise #7564 priority from low to medium, so it is implemented for 1.14 release.

dleach02, 2018-09-17:
I agree. 

Although I do think we should fix the python script file for situations where the dts may have the led0 type of approach.

David

erwango, 2018-09-17:
> Although I do think we should fix the python script file for situations where the dts may have the led0 type of approach.

Sure

galak, 2020-02-06:
@dleach02 is this still an issue?  I went through the DT warnings and feel like I didn't see any of these cases.

galak, 2021-02-04:
@dleach02 I going to close this as I think its no longer relevant.  Feel free to re-open if you feel otherwise.

closed 2021-02-04 by galak