zephyrproject-rtos/zephyr #9884: MISRA-C Check return value of a non-void function

ceolin, 2018-09-10:
* The majority of hits inside the kernel are related to the printk() function call. It returns the number of characters printed. There are no places in Zephyr where its return value is actually used for anything, so it's safe to just change the prototype so it's a void function instead.
* Other cases include calls to functions such as memcpy().  This function merely returns the destination pointer, so code in Zephyr rarely use its return value.
* For functions which the return value should be checked, mark the function with ``__attribute__((warn_unused_return))`` and fix all cases where the return value isn't used
* For functions which the return value don't need to be checked, consider changing the function to a return void instead
* For cases where the return value can be ignored, cast the function call to (void)

Part of #9882

metapsychologe, 2018-09-10:
According to the return values of memcpy I really doubt that it's a good idea to disagree with the C standard.
I agree that they are not really useful though.

koffes, 2020-01-21:
Can such a test be added to Checkpatch?

closed 2021-05-28 by nashif