zephyrproject-rtos/zephyr #9893: MISRA C Review switch statement usage

ceolin, 2018-09-10:
To avoid bugs caused by wrong usage of `switch` statement, a serie of good practices must be taken.

* Every single switch clause must have an unconditional break, empty fall is allowed though. e.g:
```
switch (x) {
   case 'a':  /* allowed */
   case 'b':
      do_something();
      break;
  case 'c':
      do_something_else(); /*not allowed */
  case 'd':
      break;
  default:
      break;
}
```
* Every switch must have a default label and it must appear either as first or last label.
* A switch must have at least two clauses

part of #9552

ceolin, 2018-09-11:
#9927

ceolin, 2018-09-26:
rules 16.1, 16.2, 16.3, 16.4, 16.5, 16.6 and 16.7

himanshujha199640, 2018-11-13:
Does this MISRA-C rule open opportunities for enabling `-Wimplicit-fallthrough` to the standard
list of warnings ?

AFAIK, Coverity produces a lot of bugs tagged "Control flow Issues" due to the fallthrough warnings. And
more often they are actually *real* bugs. 

OTOH if we expect that the behavior is desired/intentional we mark the switch case as "fall through" stringwhich compiler detects/reads to silence the warnings.

At least in the linux kernel community, we are soon (maybe next merge window) going to enable it and there has been continuous
effort in the community by Gustavo A.R. Silva in that direction.

More insights: https://developers.redhat.com/blog/2017/03/10/wimplicit-fallthrough-in-gcc-7/

And I see some usage already in Zephyr in various subsystems:
```
himanshu@himanshu-Vostro-3559:~/zephyr$ git grep -n "fall through"
arch/arc/core/atomic.S:61:      /* fall through into atomic_set */
arch/arc/core/atomic.S:134:     /* fall through into atomic_add */
arch/arc/core/atomic.S:195:     /* fall through into atomic_sub */
arch/arc/core/fast_irq.S:276:   /* fall through */
arch/arc/core/regular_irq.S:135:        /* cached thread to run is in r0, fall through */
arch/arc/core/regular_irq.S:176:        /* fall through */
arch/arc/core/regular_irq.S:213:        /* fall through to rtie instruction */
arch/arc/core/regular_irq.S:217:        /* fall through to rtie instruction */
arch/arc/core/swap.S:152:       /* fall through to _swap_return_from_coop */
arch/x86/core/crt0.S:402:       /* fall through to handle the remaining double words (32-bit chunks) */
arch/x86/core/intstub.S:190:    /* fall through to nested case */
arch/x86/core/intstub.S:334:    /* fall through to 'nestedInterrupt' */
arch/x86/core/intstub.S:421:    /* fall through to _SpuriousIntHandler */
arch/x86/core/intstub.S:508:                     * a block, in which case we just fall through
arch/x86/core/swap.S:258:       /* fall through to 'floatSaveDone' */
arch/x86/core/swap.S:289:       /* fall through to 'floatRestoreDone' */
drivers/i2c/i2c_dw.c:328:               /* fall through */
drivers/i2c/i2c_dw.c:551:               /* fall through */
drivers/i2s/i2s_cavs.c:154:             /* fall through */
ext/hal/st/stm32cube/stm32l4xx/drivers/src/stm32l4xx_hal_hash.c:2141:    - or fall through from the Step 1 processing carried out hereabove */
ext/hal/st/stm32cube/stm32l4xx/drivers/src/stm32l4xx_hal_hash.c:2188:    - or fall through from the Step 2 processing carried out hereabove */
ext/lib/crypto/mbedtls/library/ssl_tls.c:7159:     * in process. If yes, perform/continue, and fall through
ext/lib/encoding/tinycbor/src/cborparser.c:242:            /* fall through */
samples/net/zperf/src/zperf_tcp_receiver.c:70:          /* fall through */
subsys/bluetooth/controller/ll_sw/ctrl.c:1854:          /* fall through */
subsys/bluetooth/host/hci_core.c:1707:          /* fall through */
subsys/bluetooth/host/hci_core.c:1718:          /* fall through */
subsys/bluetooth/host/mesh/cfg_srv.c:2234:              /* fall through */
subsys/bluetooth/host/mesh/lpn.c:747:           /* fall through */
subsys/shell/shell.c:879:                               /* fall through */
subsys/shell/shell.c:888:                               /* fall through */
subsys/shell/shell.c:897:                               /* fall through */
subsys/shell/shell_log_backend.c:127:           /* fall through */
tests/kernel/mbox/mbox_api/src/test_mbox_api.c:121:     /*fall through*/
tests/kernel/mbox/mbox_api/src/test_mbox_api.c:145:     /*fall through*/
tests/kernel/mbox/mbox_api/src/test_mbox_api.c:316:     /*fall through*/
tests/kernel/mbox/mbox_api/src/test_mbox_api.c:346:     /*fall through*/
```

ceolin, 2018-11-13:
We can, enable the warning is a good thing once we fixed it because will avoid errors be introduced. Regarding `__attribute__ ((fallthrough)); ` I particularly a big fan. We want to be, as much compiler agnostic, as possible. Also this will still need a deviation. That's being said, we can discuss it more.

ceolin, 2018-11-13:
btw, we can turn it on once we have fixed the issues to avoid break CI

ceolin, 2020-09-09:
https://github.com/zephyrproject-rtos/zephyr/pull/27752

closed 2020-09-09 by ceolin