zephyrproject-rtos/zephyr #9898: Zephyr 1.14 Checlist

nashif, 2018-09-11:
Add items as required
--------
- [x] Major enhancements:

- [x] Pre RC1 Steps
   - [x] Verify that all external components and external dependencies are up to date, for example 
      - mbedTLS
      - tinycrypt
      - FatFS
   - [x] Check known vulnerabilities, and fix exploitable vulnerabilities or verify them as unexploitable.

See https://github.com/zephyrproject-rtos/zephyr/wiki/Release-procedure for tagging details

--------
- [x] Release Notes WIP
  - [x] Create draft doc/release-notes-1.14.rst with feature summary from https://www.zephyrproject.org/developers/#releases-overview
  - [x] Update doc/release-notes-1.14.rst details sections (relative to 1.13.0 release) as important issues are addressed (subsystem owners)
--------
- [x] Release data
  - [x] Update version in VERSION
  - [x] Run sanitycheck with --all --release and checkin resulting file scripts/sanity_chk/sanity_last_release.csv
--------
- [ ] Finalize Release Notes
  - [x] Add list of GitHub issues (spell check issue list and fix in GitHub as needed)
  - [x] Review doc/release-notes-1.14.rst overview summary and details sections (see Sign-off below)
  - [x] Update doc/release-notes.rst (add reference to 1.14 notes)
  - [x] Update doc/index.rst (add reference to doc/1.14.0)
  - [x] Update doc/conf.py (add version to pick list)
  - [x] Update doc/LICENSING.rst for new components that do not have a license
  - [x] Create release and add notes on https://github.com/zephyrproject-rtos/zephyr/releases
  - [x] Final tagged release as zephyr-v1.14.0 (with release notes)
  - [ ] Update https://www.zephyrproject.org/developers/#releases-overview
  - [ ] Update https://www.zephyrproject.org/developers/#downloads (and git checkout version example)
  - [x] Update doc generation (in future should be automatic based on tag) ping @nashif
  - [ ] Email devel@lists.zephyrproject.org & announce@lists.zephyrproject.org lists about release
--------
- [x]  Final Release Notes Sign-off
  - Subsystems
    - [x] Networking - @jukkar 
    - [x] Bluetooth - @jhedberg , @carlescufi 
  - Architectures
    - [x] ARC - @ruuddw 
    - [x] ARM - @galak, @MaureenHelm 
    - [x] X86 - @nashif, @andrewboie 
    - [x] Xtensa, RISCV32, NIOS-II - @nashif, @andrewboie 
  - [x] Security - @agross-linaro 
--------
- [ ] After Release
  - [x] Update patchlevel in Makefile (PATCHLEVEL = 99)
  - [ ] Marketing Blog for announcing release (https://www.zephyrproject.org/blog)
  - [x] Create branch for 1.14.x releases, update github protections for branch
  - [ ] Create checklist issue for next release
  - [ ] Add a comment to this issue linking to the next release checklist issue
  - [ ] Update https://www.zephyrproject.org with latest stats from Github

galak, 2019-04-16:
#15494

galak, 2019-04-17:
@Thea-Aldrich added you for the "marketing blog".

Thea-Aldrich, 2019-04-17:
Thanks! We have 2 pieces that will go out this week. We have a formal press release that will be posted to the website as soon as it hits the wire. That is scheduled for today IIRC. That will also trigger the official social media promotion.

Friday we will be posting a second blog that includes additional quotes, more details from the release notes and a reminder to get entries in for the bug fixing contest.

closed 2019-05-23 by nashif