zephyrproject-rtos/zephyr #9922: Networking: Neighbour Discovery may be breaking an ongoing UDP Transmission

KillerLink, 2018-09-11:
Hello,
im currently in the proccess of writing a platform port to the Texas Instruments CC2538. I have working UART, GPIO and IEEE802.15.4 RF (as far as i can tell currently). I'd like to point out that by no means my code is perfect and the error offcourse could reside within my code. I am on current code base i got by "git pull" at the time of writting this issue.

To the problem:
I am using IPV6/UDP echo_client and echo_server to test my implementation for functionality. The transmission works fine with very minimal package loss (close to none) except for a periodic event which causes me to lose exactly one package. Approximately every 27 seconds there is an event that causes exactly one UDP packet to be "seemingly" sent by the echo_client but no response is recieved. I now have used a sniffer to track down the problem and noticed that the UDP packet in question is apparently not sent entirely over the air, only the first IEEE802.15.4 fragment of the UDP packet is sent, but without an error notification to the echo_client application (see Note 1&2). I do realize that i am using UDP which does not make transmission guarantess, but as far as i tracked down the error, tx_packet_fragments(...) returns -5 for the UDP packet in question, but this does not lead to an error returned from the net_app_send_pkt(...) function in the echo_client. I would expect UDP to throw an error if it notices that it could not even transmitt an outgoing package completly. 
My next observation with Wireshark was that always when the problem occurs, the first IEEE802.15.4 fragment of the UDP packet is followed by a neighbour solicitation by the echo_server, a neighbour advertisement by the echo_client, a neighbour solicitation from the echo_client and a neighbour advertisement by the echo_server (see attached screenshots of wireshark) (see Note 1&2).

I also attached a log with some network debugging enabled which include such failed transmissions.
LinkLayer/IPv6 addresses ending in a 1 are my echo_server.
LinkLayer/IPv6 addresses ending in a 2 are my echo_client.

My questions are:
1) Is it wrong to assume that a client application should be notified in some way if an outgoing udp transmission failed (e.g.: via the return value of net_app_send_pkt)?
2) Is the behaviour, that the incoming Neighbour Discovery traffic completly breaks the current UDP transmission correct or might there be a bug hidding? I do miss the knowledge (about IPv6 and UDP in depth) to effectively call it a bug.
3) It seems that the first IEEE802.15.4 fragment of the UDP packet sent by the echo_client triggers the echo_server to start the visible Neighbour Discovery traffic, could you hint me to why that is? I would assume since there is constant traffic (>1 UDP packet per second) that reachability timers _should_ not be the issue since reachability could be interfered by the incoming/outgoing packages?

*Note 1:
 apparently i can get some sporadic runs where it mostly *just works fine* which seems to be especially after i let the echo_server run for longer time and had reset the echo_client a few times.
 
*Note 2:
 apparently the Neighbour Discovery is not always triggered after the first IEEE802.15.4 fragment of the UDP packet, i have seen runs where it seems to happen reliably after the first fragment, some where it seems just inbetween a UDP packet sent by the echo_client.

[client.log](https://github.com/zephyrproject-rtos/zephyr/files/2372412/client.log)

TRACE 0:
[image]
TRACE 1:
[image]
TRACE 2:
[image]

KillerLink, 2018-09-11:
Here is my .conf file i use for my echo_client, plus some debug options i turned on by hand.
echo_server is almost identical to that.

```
CONFIG_MAIN_STACK_SIZE=2048

# Generic networking options
CONFIG_NETWORKING=y
CONFIG_NET_IPV6=y
CONFIG_NET_IPV4=n
CONFIG_NET_UDP=y
CONFIG_NET_TCP=n

# Kernel options 
CONFIG_ENTROPY_GENERATOR=y
CONFIG_TEST_RANDOM_GENERATOR=y
CONFIG_SYS_LOG_NET_BUF_LEVEL=2

# Logging
CONFIG_NET_LOG=y
CONFIG_NET_BUF_LOG=y
CONFIG_SYS_LOG_NET_LEVEL=4
CONFIG_SYS_LOG_SHOW_COLOR=y
CONFIG_INIT_STACKS=y
CONFIG_PRINTK=y
#CONFIG_NET_DEBUG_L2_IEEE802154=y
#CONFIG_SYS_LOG_IEEE802154_DRIVER_LEVEL=1
#CONFIG_SYS_LOG_DEFAULT_LEVEL=4
#CONFIG_SYS_LOG_OVERRIDE_LEVEL=4

# Network buffers
CONFIG_NET_STATISTICS=y
CONFIG_NET_PKT_RX_COUNT=28
CONFIG_NET_PKT_TX_COUNT=28
CONFIG_NET_BUF_RX_COUNT=28
CONFIG_NET_BUF_TX_COUNT=28
CONFIG_NET_CONTEXT_NET_PKT_POOL=y

# IP address options
CONFIG_NET_IF_UNICAST_IPV6_ADDR_COUNT=6
CONFIG_NET_IF_MCAST_IPV6_ADDR_COUNT=6
CONFIG_NET_MAX_CONTEXTS=10

# L2 Options
CONFIG_NET_L2_IEEE802154=y
CONFIG_NET_L2_IEEE802154_FRAGMENT_REASS_CACHE_SIZE=20
CONFIG_NET_L2_IEEE802154_REASSEMBLY_TIMEOUT=10

# Network shell
CONFIG_NET_SHELL=n

# Network application options and configuration
CONFIG_NET_APP=y
CONFIG_NET_APP_CLIENT=y
CONFIG_NET_DEBUG_APP=n
CONFIG_NET_CONFIG_SETTINGS=y
CONFIG_NET_CONFIG_NEED_IPV6=y
CONFIG_NET_CONFIG_NEED_IPV4=n
CONFIG_NET_CONFIG_MY_IPV6_ADDR="2001:db8::2"
CONFIG_NET_CONFIG_PEER_IPV6_ADDR="2001:db8::1"
CONFIG_NET_CONFIG_MY_IPV4_ADDR=""
CONFIG_NET_CONFIG_PEER_IPV4_ADDR=""

CONFIG_IEEE802154_TI_CC2538_RF=y
CONFIG_IEEE802154_TI_CC2538_RF_RANDOM_MAC_BYTES=0
CONFIG_IEEE802154_TI_CC2538_RF_MAC0=0x00
CONFIG_IEEE802154_TI_CC2538_RF_MAC1=0x12
CONFIG_IEEE802154_TI_CC2538_RF_MAC2=0x4b
CONFIG_IEEE802154_TI_CC2538_RF_MAC3=0x00
CONFIG_IEEE802154_TI_CC2538_RF_MAC4=0x00
CONFIG_IEEE802154_TI_CC2538_RF_MAC5=0x00
CONFIG_IEEE802154_TI_CC2538_RF_MAC6=0x00
CONFIG_IEEE802154_TI_CC2538_RF_MAC7=0x02

CONFIG_NET_CONFIG_IEEE802154_DEV_NAME="ti_cc2538_rf_radio"
CONFIG_IEEE802154_TI_CC2538_RF_DRV_NAME="ti_cc2538_rf_radio"
```

iotsys, 2018-09-12:
Hello, when I test LTE-Modem, when 2 UDPs are enabled at the same time, the length of the transmitted data may be 0. I have not found the reason for this problem, but it may be caused by the same reason as you.

KillerLink, 2018-09-17:
@iotsys I did not yet see that the length of data to be sent would be 0, where/how did you observe this? Anyways, it sounds like a different issue to me, i am not sure though.

jukkar, 2018-10-19:
Sorry about late reply, this issue got lost in to the sea of other issues.

> * Is it wrong to assume that a client application should be notified in some way if an outgoing udp transmission failed (e.g.: via the return value of net_app_send_pkt)?

If the IEEE 802.15.4 device driver TX returns error, then for example in ieee802154_radio_csma_ca.c line 69 should do retries and return error code to net_if.c line 173 (status = api->send()), which would then be given to application / net_app by net_context_send_cb() if there was send_cb defined when packet was sent. So in theory, the return code should be returned to the application, but there are many if's here where things could go wrong.

>  * Is the behaviour, that the incoming Neighbour Discovery traffic completly breaks the current UDP transmission correct or might there be a bug hidding? I do miss the knowledge (about IPv6 and UDP in depth) to effectively call it a bug.

No, incoming ND should not cause any issues if there are still enough network buffers available when sending normal UDP traffic.

> * It seems that the first IEEE802.15.4 fragment of the UDP packet sent by the echo_client triggers the echo_server to start the visible Neighbour Discovery traffic, could you hint me to why that is? [...]

This is the Neighbor Unreachability Detection algorithm that does things every 30 seconds. See for example https://sites.google.com/site/amitsciscozone/home/important-tips/ipv6/neighbor-discovery-in-ipv6 for some details. Also https://tools.ietf.org/html/rfc4861#page-68 describes this behavior.

KillerLink, 2018-10-19:
No problem!

If I understood you correctly, that means my application should recieve an error from the network stack IF my phsical layer driver delivered error(s) (on all retries) to the network stack. I will check again exactly whether or not my driver does this correctly. It might be possible that the ChannelClearAsessment is to insensitive.

Regarding "enough network buffers", the default configuration uses around 50 buffers while my configuration uses only 28 ( -> 99% ram usage), is this "enough" for the echo client/server? Or might there be any other configuration issue you can think of from the .conf i posted above?
( My thought was that since only one packet is on air at a time, with around ~1500 maximum length, 15 buffers should be enough for that one packet plus some overhead in every fragment, and some more for those periodic Neighbour Discoveries.)

I see that there are recent changes to the network stack, i will pull them next week, test again and report back with my findings.

jukkar, 2018-10-19:
> Regarding "enough network buffers", the default configuration uses around 50 buffers while my configuration uses only 28 ( -> 99% ram usage), is this "enough" for the echo client/server? Or might there be any other configuration issue you can think of from the .conf i posted above?

You buffer count sounds very reasonable, there should be no issues in that side.

jukkar, 2018-10-19:
> If I understood you correctly, that means my application should recieve an error from the network stack IF my phsical layer driver delivered error(s) (on all retries) to the network stack. I will [...]

Yes, the return code from driver should propagate up in the stack. But of course bugs are possible here, I tried to follow the path from driver to application but might have missed something.

Any possibility for you to send your driver to upstream for a review?

KillerLink, 2018-11-17:
So with new knowledge gained from a lot of trial and error, i rewrote my ieee802154 driver to the best of my knowledge, built upon the current state of zephyr. I now can not reproduce the errors / bug i previously did, which leads me to the assumption that it either has been fixed on the network stack side or on my driver side. 
The propagation of the error from the stack upward works as expected now, i noticed that the send_cb mentioned earlier initially was not present in the echo_client.

Im still working on getting the driver upstream, but i haven't finished cleaning up the amount of hacks i created while debugging. 

I am closing this bug because i could not reproduce the specific bug i described anymore.

closed 2018-11-17 by KillerLink