zephyrproject-rtos/zephyr #9924: Segger Systemview + newlib does not compile

vanwinkeljan, 2018-09-11:
In case both Segger Systemview (CONFIG_SEGER_SYSTEMVIEW) and newlib (CONFIG_NEWLIB_LIBC) are selected in a project, the project does not compile as _thread_entry is undeclared.

vanwinkeljan, 2018-09-11:
In case it is of any help:
```
In file included from .../zephyr/kernel/include/kernel_structs.h:182:0,
                 from .../zephyr/subsys/debug/tracing/include/tracing_sysview.h:9,
                 from .../zephyr/include/tracing.h:29,
                 from .../zephyr/include/kernel.h:4981,
                 from .../zephyr/kernel/include/kernel_internal.h:17,
                 from .../zephyr/lib/libc/newlib/libc-hooks.c:13:
.../zephyr/arch/arm/include/kernel_arch_func.h: In function __arch_switch_to_main_thread_:
.../zephyr/arch/arm/include/kernel_arch_func.h:104:21: error: __thread_entry_ undeclared (first use in this function)
     "r"(_main), "r"(_thread_entry),
```

leodido99, 2018-10-30:
I'm having the same problem. Is there a quick fix to this?

vanwinkeljan, 2019-01-09:
not reproducible anymore, closing issue

closed 2019-01-09 by vanwinkeljan

vanwinkeljan, 2019-01-23:
@MrPointer could you share how you reproduced?
I tried with hello world sample by enabling CONFIG_SEGER_SYSTEMVIEW and CONFIG_NEWLIB_LIBC

MrPointer, 2019-01-24:
Well first I tried it on a personal project, but now I've tried it on the **philosophers** example as recommended in the docs, and it still fails.
Besides `CONFIG_SEGER_SYSTEMVIEW` and `CONFIG_NEWLIB_LIBC`, I've set `CONFIG_HAS_SEGGER_RTT` as well (seems obvious but might be important to point).

vanwinkeljan, 2019-01-24:
Just tested with philosophers (added flags mentioned above) sample on both reel_board and nrf52840_pca10056  boards, and still cant reproduce.

MrPointer, 2019-01-25:
Hmm... I also tested it with the **nrf52840_pca10056** board 😐 
Maybe it's worth comparing the toolchains each of us use? Personally I use the pretty "old" `gcc-arm-mbed-2017-q4` toolchain. Besides, "my" **newlib** is the one that comes built-in with **Zephyr** ***1.13.0***

vanwinkeljan, 2019-01-25:
@MrPointer  Fyi, I'm using the latest Zephyr SDK (0.9.5)

MrPointer, 2019-01-27:
Just tested it using the latest gcc-arm mbed's toolchain (version 2018 q4), still no luck.
Will update when I'll get the chance to test using the SDK.

MrPointer, 2019-02-17:
**Update:** After upgrading to the latest SDK (0.9.5), the issue ***still*** persists.