zephyrproject-rtos/zephyr #9926: samples/net/sockets/echo_server: No docs for testing TLS mode

pfalcon, 2018-09-11:
Subj. Nothing in the docs. I tried:

~~~
openssl s_client -host 192.0.2.1 -port 4242
~~~

Got:

~~~
**** Booting Zephyr OS zephyr-v1.13.0-53-g1e58bc1d03 *****
Zephyr Shell, Zephyr version: 1.13.99
Type 'help' for a list of available commands
shell> [echo-server] [INF] init_app: Run echo server
[echo-server] [INF] process_tcp: Waiting for TCP connection (IPv6)...
[echo-server] [INF] process_tcp: Waiting for TCP connection (IPv4)...
[echo-server] [INF] process_udp: Waiting for UDP packets (IPv6)...
[echo-server] [INF] process_udp: Waiting for UDP packets (IPv4)...
[echo-server] [ERR] process_tcp: Error in accept (IPv6): 53 - stopping server
[echo-server] [INF] main: Stopping...
~~~

Not good.

pfalcon, 2018-09-11:
P.S. Actually, there's even no docs how to build TLS mode...

pfalcon, 2018-09-11:
> And well, the above issue happens when built against min libc.

Ah, not even that - it happened during *that run*. Rebuild/restart - openssl s_client works.

pfalcon, 2018-09-11:
So, I'll include switching these to minlibc to https://github.com/zephyrproject-rtos/zephyr/pull/9923

nashif, 2018-09-11:
who are you talking to?

pfalcon, 2018-09-11:
To the assignee of the ticket, the author of this sample.

rlubos, 2018-09-12:
Sorry for missing docs, I have in in my schedule, just didn't come to it yet. It will happen though.

> P.P.S. And having boards/qemu_x86.conf is counter-intuitive/productive/etc.

So where would you see this config then? This is approach suggested in review of #8342 and followed by #9033 as a way to solve the problem of multiple, almost identical config files. If you have other/better propsal to make this simple, please share with us instead of writing daunting comments.

> So, I'll include switching these to minlibc to #9923

I appreciate that.

Anyway, thanks for leaving the ticket, so it won't be forgotten.

pfalcon, 2018-09-13:
> > P.P.S. And having boards/qemu_x86.conf is counter-intuitive/productive/etc.

> So where would you see this config then?

So, the idea which I hoped the mainline by now and in everyone's loop, is that generic networking functionality should not require board-specific configuration. I.e. if a board has some IP-compatible networking builtin, the the board config should enable/configure it by default. Yes, if people need to use non-default choice, or use external hardware, they'd need to have board-specific (re)configuration, by default should just work.

That's how my samples in samples/net/sockets/ are written.

And let's look inside your boards/qemu_x86.conf:
~~~
CONFIG_NET_UDP=y
CONFIG_NET_TCP=y
CONFIG_NET_IPV6=y
CONFIG_NET_IPV4=y
~~~ 

How is that qemu_x86 specific? Please just merge those opts into the main prj.conf and remove that file.

rlubos, 2018-09-13:
> How is that qemu_x86 specific? Please just merge those opts into the main prj.conf and remove that file

When I was suggested to introduce this `boards/${BOARD}.conf` pattern, I've took a look what differentiated boards in the original echo samples. What was different beside hardware specific stuff was that some boards did not enable IPv4, others didn't run TCP etc. (Perhaps memory constraints? Techonology limitations?). That's why I put those configs into separate file.

Of course, we could enable everyting by default, and then have other boards override these values, perhaps it is a better way to do that. But again, if we want to enforce this new pattern of adding configs, I think that it is good to have a reference of how board-specific configuration should be added. Otherwise we'll end up with people submitting configs in an old way (because it's simpler/faster to just copy a config and modify what's needed).

pfalcon, 2018-09-13:
> When I was suggested to introduce this boards/${BOARD}.conf pattern, I've took a look what differentiated boards in the original echo samples.

Let me start reading here, write the response to this, then continue reading ;-).

The "original echo samples" are messy config with too much magic which everybody is afraid to touch to not break some magic somebody else may be using (for testing, and testing is important). Rewrite, especially with sockets, gives an opportunity to leave that baggage behind. (And I overlooked that during the original review, yeah.)

pfalcon, 2018-09-13:
> memory constraints? 

Would be handled on board level. Like, if it really can't use TCP or IPv6, should be force-disable on board level. But I doubt we have choices like that, and if some board is really on an edge - yes, that can be handled by board-specific config.

> Techonology limitations?

Should be handled automagically too. E.g. 6lowpan devices can't work with IPv4. So, it can be either: 1) IPv4 enabled, so what; or 2) just be force-disabled. I believe we're at 2) already. Users would get warning of conflicting CONFIG_IPV4 assignment after recent @ulfalizer work, but that's price of magic.

 >  it is good to have a reference of how board-specific configuration should be added

Sure. It's just socket and IP networking in general examples aren't the best place for that. We surely have other examples with board/*.conf ?

> Otherwise we'll end up with people submitting configs in an old way (because it's simpler/faster to just copy a config and modify what's needed).

No, we'll -1 that ;-). There should be single prj.conf per sample which just works. Anything else is rare exception which should have good justification ;-).

rlubos, 2018-09-13:
Well, I don't want to argue here, as in general I agree with your point of view, it's just that this vision is idealistic and would invovle changes not only in project files but in module configurations as well (which sounds like lot of work), but yes, we should strive for that.

> We surely have other examples with board/*.conf ?

Not many actually, at the moment the echo socket samples were submitted there was one or two samples that supported this pattern. But hopefully PR's like #9033 will not be a single case, and could serve as a reference for others (still not ideal, but a good step forward).

I'm not having problem with removing boards directory from echo_client/server if there's a will to do that, at least unitl it's necessary to have it. So let's just summarize this topic in a checklist, that I could take care of once I'm done with current stuff.

rlubos, 2018-09-13:
So to summarize:

- [ ] Add TLS sockets subsystem documentation
- [ ] Update socket echo_client/echo_server samples with information on how to run TLS
- [ ] Unify socket echo_client/echo_server sample project configuration into a single file (with TLS as overlay config)

I hope that concludes updates requested in this topic?

pfalcon, 2018-09-17:
> I hope that concludes updates requested in this topic?

Yeah, sorry for brining different things under the same ticket ;-). I guess it could work as "beginning of new devel cycle brainstorming", but feel free to request moving different topics to different tickets.

> Feel free to update if I've missed something.

Looks good. I just took freedom to reorder bullet points from easier/sooner to do down to requiring more time to do.

rlubos, 2018-09-26:
I'll reopen this, there's still one point to cover.

rlubos, 2018-11-22:
Closing, as #11534 satisified the final point from the checklist.

PR #10211 (2018-09-26): samples: net: sockets: Update docs with TLS information
Update documentation of socket samples that support TLS with information
on how to enable it.

Partially fixes #9926
closed 2018-11-22 by rlubos