zephyrproject-rtos/zephyr #9932: Invalid documationtion link in README.rst

kestewart, 2018-09-12:
https://github.com/zephyrproject-rtos/zephyr/blob/master/README.rst has invalid link to new boards

The Zephyr kernel supports multiple architectures, including ARM Cortex-M, Intel x86, ARC, Nios II, Tensilica Xtensa, and RISC-V, and a large number of supported boards.

Supported boards link redirects to https://docs.zephyrproject.org/latest/404.html

nashif, 2018-09-12:
removed the redirects after founding issues, fixed links now and will revisit the redirects later.

dbkinder, 2018-09-12:
To clarify, the "problem" is that we [moved the latest documentation down into a /latest folder](https://github.com/zephyrproject-rtos/zephyr/issues/5524) (a sibling folder to the tagged release documentation such as /1.13.0 and 1.12.0).  Now, any deep link such as  https://docs.zephyrproject.org/boards/boards.html that does not include the doc version folder name (or /latest) will return a 404 error.  We can fix links within the Zephyr documentation, but not incoming links from other sites.  Over time, folks will update their links, but until then, we're likely to get a barrage of 404 errors from google searches and links from external sites.  

For example, doing a [google search for "zephyr usb support"](https://www.google.com/search?q=zephyr+usb+support)   returns a link to http://docs.zephyrproject.org/subsystems/usb/usb.html which will get a 404 error.

nashif, 2018-09-12:
not really concerned about google, it will update its database soon enough.

btw, clicking links from google that lead to 404 have this message:

`"Sorry, the page you requested: https://www.google.com/ was not found on this site."`

:)

Need to fix this.

PR #9942 (2018-09-12): docs: fixed documenation pointers
Fixed URL to documentation, now latest docs are under /latest/..
Fixes #9932
commit 81b27211 (2018-09-12): docs: fixed documenation pointers

Fixed URL to documentation, now latest docs are under /latest/..
Fixes #9932
closed 2018-09-12 by nashif