zephyrproject-rtos/zephyr #9935: sockets: Connect always binds the context to the default interface

leodido99, 2018-09-12:
I'm working with sockets on v1.13 and I have a case where I have two interfaces and I'm trying to open a socket on the index 1 interface (i.e. not the default one). When doing connect and providing the remote address and port to a TCP server connected on my interface Zephyr sends the SYN on the wrong interface.

The call path goes like this:
connect
_impl_zsock_connect
net_context_connect
bind_default
net_context_bind

The problem is that bind_default provides an unspecified address to net_context_bind, so naturally net_context_bind takes the default interface. This is done in ALL cases because bind_default defines a local address and specifically sets it to unspecified instead of using the remote address to determine the local address and then the local interface.

I did an ugly hack to test that it works and it manages to connect now since it's sending SYN to the appropriate interface. Since I didn't want to heavily modify the code I just let bind_default do its job then just override local address and interface in the context by doing the following:

		/* Find the appropriate local address */
		laddr = net_if_ipv6_select_src_addr(NULL, &addr6->sin6_addr);
		ifaddr = net_if_ipv6_addr_lookup(laddr, &iface);
		if (!ifaddr) {
			SYS_LOG_ERR("Cannot determine network interface");
			return -ENOENT;
		}

		/* Set context */
		net_context_set_iface(context, iface);
		net_sin6_ptr(&context->local)->sin6_addr = &ifaddr->address.in6_addr;

I'm not sure net_context_connect is the correct place to do this honestly. Net app for exemple does this in _net_app_set_local_addr during net_app_init_tcp_client.

Was this an oversight or did I miss something?

leodido99, 2018-09-12:
After investigating more I modified _impl_zsock_connect to bind the context in case it is not already done since I think this is the correct way to do it.

Let me know what you think.

```
int _impl_zsock_connect(int sock, const struct sockaddr *addr,
			socklen_t addrlen)
{
	struct net_context *ctx = INT_TO_POINTER(sock);
	struct sockaddr *local_addr = NULL;
	socklen_t local_addr_len = 0;
#if defined(CONFIG_NET_IPV6)
	struct sockaddr_in6 local_addr6 = {
		.sin6_family = AF_INET6,
		.sin6_port = 0,
	};
#endif
#if defined(CONFIG_NET_IPV4)
	struct sockaddr_in local_addr4 = {
		.sin_family = AF_INET,
		.sin_port = 0,
	};
#endif

	/* If context is not already bound do it now */
	if (!ctx->conn_handler) {
		if (addr->sa_family == AF_INET6) {
#if defined(CONFIG_NET_IPV6)
			const struct in6_addr *laddr6;

			laddr6 = net_if_ipv6_select_src_addr(NULL,
					&net_sin6(addr)->sin6_addr);
			if (!laddr6) {
				SYS_LOG_ERR("Cannot determine local address");
				return -ENOENT;
			}

			local_addr = (struct sockaddr *)&local_addr6;
			local_addr_len = sizeof(struct sockaddr_in6);
			net_ipaddr_copy(&local_addr6.sin6_addr, laddr6);
#endif
		} else {
#if defined(CONFIG_NET_IPV4)
			const struct in_addr *laddr4;

			laddr4 = net_if_ipv4_select_src_addr(NULL,
					&net_sin(addr)->sin_addr);
			if (!laddr4) {
				SYS_LOG_ERR("Cannot determine local address");
				return -ENOENT;
			}

			local_addr = (struct sockaddr *)&local_addr4;
			local_addr_len = sizeof(struct sockaddr_in);
			net_ipaddr_copy(&local_addr4.sin_addr, laddr4);
#endif
		}

		SET_ERRNO(net_context_bind(ctx, local_addr, local_addr_len));
	}

	SET_ERRNO(net_context_connect(ctx, addr, addrlen, NULL, K_FOREVER,
				      NULL));
	SET_ERRNO(net_context_recv(ctx, zsock_received_cb, K_NO_WAIT, ctx->user_data));

	return 0;
}
```

pfalcon, 2018-09-12:
> The problem is that bind_default provides an unspecified address to net_context_bind, so naturally net_context_bind takes the default interface. 

@leodido99 : A natural question then - why don't you bind to the right interface (address) then? Can you provide POSIX references that it should work like you describe?

> This is done in ALL cases because bind_default defines a local address and specifically sets it to unspecified instead of using the remote address to determine the local address and then the local interface.

@jukkar: Otherwise, that's finally (like, I waited for it) provides an answer why you bothered with those address selection routines in net_app, while I didn't. But then net_app was a wrong place for them, they should be done on net_context level. (Writing out of top of my head, may be missing/mixing something.)

leodido99, 2018-09-12:
As I commented, I understood after a while the purpose of net_context_bind calling bind_default that's why I moved my modification I did to socket.c which is I believe the correct place.
I'm trying to understand how you are supposed to use sockets in Zephyr because the behaviour is not the same than on Linux where you don't need to bind a client socket.

leodido99, 2018-09-12:
Understood, no basically I was very naive and just tried to use sockets as I would on Linux and found out that it wasn't working as I expected :) I have no POSIX references that suggest this should be the way.

jukkar, 2018-09-13:
Proposal that fixes this can be found at #9963, @leodido99 please test this if possible.