zephyrproject-rtos/zephyr #9944: sockets: Connect doesn't send new SYN in case the first connection attempt fails

leodido99, 2018-09-12:
In the current implementation of the sockets, when doing a connect call, if the remote doesn't reply with SYN, ACK then the call will never return and no SYN will ever be re-sent.

This is because _impl_zsock_connect calls net_context_connect with a timeout value of K_FOREVER that means that after sending the SYN the function will wait forever for an answer, if none comes then we are stuck.
I think the comment of net_context_connect would benefit in being clarified because it reads like this:
```
 * @param timeout    The timeout value for the connection. Possible values:
 *                   * K_NO_WAIT: this function will return immediately,
 *                   * K_FOREVER: this function will block until the
 *                                      connection is established,
 *                   * >0: this function will wait the specified ms.
```

It is blocking until the connection is established, however it won't resend any SYN.
I personally made a modification so I can work with them but it changes the behaviours of the sockets to not being blocking. In Linux, correct me if I'm wrong, but by default sockets are blocking and on connect, the SYN will actually be resent until an answer is received at which point it will return.

Here is my modification, I had to do the net_conn_unregister otherwise, on the retries net_tcp_register complains a connection handler is already registered.
```
	err = net_context_connect(ctx, addr, addrlen, NULL, TCP_CONNECT_TIMEOUT, NULL);
	if (err < 0) {
		/* Cannot connect, we need to unregister the
		 * connection handler, otherwise net_tcp_register
		 * will complain during the next connection attempt
		 */
		if (net_conn_unregister(ctx->conn_handler) < 0) {
			SYS_LOG_WRN("Cannot unregister socket connection handler");
		}
		errno = -err;
		return -1;
	}
```
With this modification I can see on Wireshark that the new SYN are sent at every call, the only doubt I have is regarding the local port number, which stays the same on all retries. In net app I noticed the port is forced to be changed between each retries however I looked around in RFC etc... and didn't find anything that said this is mandatory so I left it like that.
In my opinion, to make a proper implementation you'd have to check the SOCK_NONBLOCK flag and then either do a loop in connect or let user call connect each time in case of non blocking.

pfalcon, 2018-12-03:
Now all hopes are on https://github.com/zephyrproject-rtos/zephyr/pull/11443 to resolve this issue.

laperie, 2019-03-11:
This will be not fixed in an existing TCP stack as it requires quite a re-engineering of the existing source.

raveslave, 2019-04-03:
what's a good workaround for this?

karthikprabhu17, 2019-05-03:
@leodido99 : Can you tell us in which file did you make your modification?

lbise, 2019-06-25:
> @leodido99 : Can you tell us in which file did you make your modification?

I kinda missed this question. The problem is located in subsys/net/lib/sockets/sockets.c in function zsock_connect_ctx. Specifically this call : https://github.com/zephyrproject-rtos/zephyr/blob/a30162be352646c4d9dc2cc9a0cf1dfd9911a476/subsys/net/lib/sockets/sockets.c#L328
Here K_FOREVER is passed to net_context_connect which eventually leads to a call to net_tcp_connect which takes a semaphore using this timeout. The semaphore is released by tcp_synack_received so if we never receive synack then we are stuck forever.

jukkar, 2019-08-15:
We have a timeout now for connect call, I think that can be considered as a workaround for this issue so closing this one.

pfalcon, 2019-08-15:
> We have a timeout now for connect call, I think that can be considered as a workaround

As a workaround, yes. As a solution - no. So, reopening. (Added "future" milestone, so there was less incentive to close this unfixed issue during release rush of a particular version.)

ioannisg, 2019-08-17:
Adding low, since there seems to be a workaround, at least

closed 2020-10-06 by github-actions[bot]