zephyrproject-rtos/zephyr #9947: CMake build architecture documentation

tejlmand, 2018-09-12:
Currently the CMake documentation 
https://docs.zephyrproject.org/latest/application/application.html#cmake-details
targets users starting out with the creation of the own applications.

The CMake architecture, especially on how source is organized into libraries should be documented so it is clear and structured to zephyr contributors.

Areas of importance:
- Kernel
- arch
- drivers
- subsystems
  - Bluetooth
  - Network
  - etc.
- Others

marc-hb, 2019-02-12:
This is what the above commit produces:
[image]

marc-hb, 2019-02-12:
@SebastianBoe , @tejlmand, @mped-oticon, @carlescufi  is this issue #9947 the right place to discuss this sphinx-moderncmakedomain prototype and proposal? I have a number of questions (see latest commit) and I'd like to get the discussion started but not in the wrong place. Thx!
https://lists.zephyrproject.org/g/devel/topic/29744819

SebastianBoe, 2019-02-26:
Hi, very sorry for the late reply.

A dedicated PR with an RFC or DNM tag seems to me to be the best place to discuss the prototype.

SebastianBoe, 2019-12-10:
I believe this is resolved by https://github.com/zephyrproject-rtos/zephyr/pull/20324/files , if not I'd like to know more about what needs to be documented.

marc-hb, 2019-12-10:
FWIW PoC/draft #13805 demoed how sphinx can be used to extract documentation straight from the CMake code. #20324 seems much higher level (and more important!).

tejlmand, 2020-05-15:
Re-opened, as #20324 only describes the high level architecture (which is also important).

But it does not describe  `zephyr_library_*` functions and other extension commands.
And when to use `zephyr_sources()` or `zephyr_library() + zephyr_library_sources`, and so on.

What we need is something similar to the proposal in #13805

The problem today, is that people must read: https://github.com/zephyrproject-rtos/zephyr/blob/master/cmake/extensions.cmake
in order to understand Zephyr CMake extension, which is not easy for new comers.

Also, having them documented only in a cmake file, means they will not show up when people search.
Compared to raw CMake, where a lot of good online documentation is available, as well as Q/A regarding CMake on stack overflow.
(Solving #8439 could remove or minimize the need for maintaining Zephyr CMake documentation)

@marc-hb should we reopen 13805 and continue work ?

marc-hb, 2020-05-19:
> @marc-hb should we reopen 13805 and continue work ?

@tejlmand why not, however I won't be able to personally resume work on this in the short or medium term.

marc-hb, 2024-02-12:
This is still valid and valuable IMHO