zephyrproject-rtos/zephyr #9954: /zephyr/samples/hello_world/build faild

charithawee, 2018-09-13:
I'm having a problem with building a sample hello_world code. The error message is given below. please advice. 

at the every first time, It builds with out any issues, but currently it does not work. 

MSYS ~/zephyr/samples/hello_world/build
$ cmake -GNinja -DBOARD=arduino_101 ..
-- Found PythonInterp: /usr/bin/python3.exe (found suitable version "3.6.6", minimum required is "3.4")
-- Selected BOARD arduino_101
Zephyr version: 1.13.0
Parsing Kconfig tree in /home/cweerasuriya/zephyr/Kconfig
Loading /home/cweerasuriya/zephyr/boards/x86/arduino_101/arduino_101_defconfig as base
Merging /home/cweerasuriya/zephyr/samples/hello_world/prj.conf
-- Generating zephyr/include/generated/generated_dts_board.h
arduino_101.dts_compiled: Warning (unit_address_vs_reg): /soc/virtualcom@0: node has a unit name, but no reg property
extract_dts_includes.py: 'id' property missing in 'Intel I/O Advanced Programmable Interrupt Controller' binding. Using '<unknown id>'.
-- Cache files will be written to: /home/cweerasuriya/.cache/zephyr
-- The C compiler identification is GNU 7.3.1
-- The CXX compiler identification is GNU 7.3.1
-- The ASM compiler identification is GNU
-- Found assembler: /c/gnuarmemb/bin/arm-none-eabi-gcc.exe
-- Performing Test toolchain_is_ok
-- Performing Test toolchain_is_ok - Success
--
*******************************
*** NOTE TO KCONFIG AUTHORS ***
*******************************

The behavior of Kconfig 'default' properties in Zephyr has changed. The
earliest default with a satisfied condition is now used, instead of the
last one. This is standard Kconfig behavior.

See http://docs.zephyrproject.org/porting/board_porting.html#old-zephyr-kconfig-behavior-for-defaults.

To get rid of this note, create a file called 'hide-defaults-note' in the
Zephyr root directory. An empty file is fine.
-- Configuring done
-- Generating done
-- Build files have been written to: /home/cweerasuriya/zephyr/samples/hello_world/build

MSYS ~/zephyr/samples/hello_world/build
$ ninja
[2/104] Preparing syscall dependency handling

[8/104] Building C object zephyr/CMakeFiles/offsets.dir/arch/x86/core/offsets/offsets.c.obj
FAILED: zephyr/CMakeFiles/offsets.dir/arch/x86/core/offsets/offsets.c.obj
/c/gnuarmemb/bin/arm-none-eabi-gcc.exe -DBUILD_VERSION=v1.13.0-rc2-22-g42c19497b -DENABLE_EXTERNAL_ISR_HANDLING -DKERNEL -DQM_LAKEMONT -DSOC_SERIES=quark_se -D_FORTIFY_SOURCE=2 -D__ZEPHYR__=1 -I../../../kernel/include -I../../../arch/x86/include -I../../../arch/x86/soc/intel_quark/quark_se -I../../../arch/x86/soc/intel_quark/quark_se/include -I../../../arch/x86/soc/intel_quark/include -I../../../boards/x86/arduino_101 -I../../../include -I../../../include/drivers -Izephyr/include/generated -I../../../lib/libc/minimal/include -I../../../ext/hal/qmsi/include -I../../../ext/hal/qmsi/drivers/include -I../../../ext/hal/qmsi/soc/quark_se/include -isystem /c/gnuarmemb/lib/gcc/arm-none-eabi/7.3.1/include -isystem /c/gnuarmemb/lib/gcc/arm-none-eabi/7.3.1/include-fixed -Os -g -Wall -Wformat -Wformat-security -Wno-format-zero-length -imacros /home/cweerasuriya/zephyr/samples/hello_world/build/zephyr/include/generated/autoconf.h -ffreestanding -Wno-main -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -Wno-pointer-sign -Wno-unused-but-set-variable -fno-reorder-functions -fno-defer-pop -Werror=implicit-int -ffunction-sections -fdata-sections -msoft-float -std=c99 -MD -MT zephyr/CMakeFiles/offsets.dir/arch/x86/core/offsets/offsets.c.obj -MF zephyr/CMakeFiles/offsets.dir/arch/x86/core/offsets/offsets.c.obj.d -o zephyr/CMakeFiles/offsets.dir/arch/x86/core/offsets/offsets.c.obj   -c /home/cweerasuriya/zephyr/arch/x86/core/offsets/offsets.c
C:\msys64\tmp\ccNIJEsa.s: Assembler messages:
C:\msys64\tmp\ccNIJEsa.s:36: Error: unrecognized symbol type ""
C:\msys64\tmp\ccNIJEsa.s:42: Error: unrecognized symbol type ""
C:\msys64\tmp\ccNIJEsa.s:48: Error: unrecognized symbol type ""
C:\msys64\tmp\ccNIJEsa.s:54: Error: unrecognized symbol type ""
C:\msys64\tmp\ccNIJEsa.s:60: Error: unrecognized symbol type ""
C:\msys64\tmp\ccNIJEsa.s:66: Error: unrecognized symbol type ""
C:\msys64\tmp\ccNIJEsa.s:72: Error: unrecognized symbol type ""
C:\msys64\tmp\ccNIJEsa.s:78: Error: unrecognized symbol type ""
C:\msys64\tmp\ccNIJEsa.s:84: Error: unrecognized symbol type ""
C:\msys64\tmp\ccNIJEsa.s:90: Error: unrecognized symbol type ""
C:\msys64\tmp\ccNIJEsa.s:96: Error: unrecognized symbol type ""
C:\msys64\tmp\ccNIJEsa.s:102: Error: unrecognized symbol type ""
C:\msys64\tmp\ccNIJEsa.s:108: Error: unrecognized symbol type ""
C:\msys64\tmp\ccNIJEsa.s:114: Error: unrecognized symbol type ""
C:\msys64\tmp\ccNIJEsa.s:120: Error: unrecognized symbol type ""
C:\msys64\tmp\ccNIJEsa.s:126: Error: unrecognized symbol type ""
C:\msys64\tmp\ccNIJEsa.s:132: Error: unrecognized symbol type ""
C:\msys64\tmp\ccNIJEsa.s:138: Error: unrecognized symbol type ""
C:\msys64\tmp\ccNIJEsa.s:144: Error: unrecognized symbol type ""
C:\msys64\tmp\ccNIJEsa.s:150: Error: unrecognized symbol type ""
C:\msys64\tmp\ccNIJEsa.s:156: Error: unrecognized symbol type ""
C:\msys64\tmp\ccNIJEsa.s:162: Error: unrecognized symbol type ""
C:\msys64\tmp\ccNIJEsa.s:169: Error: unrecognized symbol type ""
C:\msys64\tmp\ccNIJEsa.s:175: Error: unrecognized symbol type ""
C:\msys64\tmp\ccNIJEsa.s:181: Error: unrecognized symbol type ""
C:\msys64\tmp\ccNIJEsa.s:187: Error: unrecognized symbol type ""
C:\msys64\tmp\ccNIJEsa.s:193: Error: unrecognized symbol type ""
C:\msys64\tmp\ccNIJEsa.s:199: Error: unrecognized symbol type ""
C:\msys64\tmp\ccNIJEsa.s:205: Error: unrecognized symbol type ""
C:\msys64\tmp\ccNIJEsa.s:211: Error: unrecognized symbol type ""
C:\msys64\tmp\ccNIJEsa.s:217: Error: unrecognized symbol type ""
C:\msys64\tmp\ccNIJEsa.s:223: Error: unrecognized symbol type ""
C:\msys64\tmp\ccNIJEsa.s:229: Error: unrecognized symbol type ""
C:\msys64\tmp\ccNIJEsa.s:235: Error: unrecognized symbol type ""
C:\msys64\tmp\ccNIJEsa.s:241: Error: unrecognized symbol type ""
C:\msys64\tmp\ccNIJEsa.s:247: Error: unrecognized symbol type ""
C:\msys64\tmp\ccNIJEsa.s:253: Error: unrecognized symbol type ""
C:\msys64\tmp\ccNIJEsa.s:259: Error: unrecognized symbol type ""
C:\msys64\tmp\ccNIJEsa.s:265: Error: unrecognized symbol type ""
C:\msys64\tmp\ccNIJEsa.s:271: Error: unrecognized symbol type ""
C:\msys64\tmp\ccNIJEsa.s:277: Error: unrecognized symbol type ""
C:\msys64\tmp\ccNIJEsa.s:283: Error: unrecognized symbol type ""
C:\msys64\tmp\ccNIJEsa.s:289: Error: unrecognized symbol type ""
C:\msys64\tmp\ccNIJEsa.s:295: Error: unrecognized symbol type ""
C:\msys64\tmp\ccNIJEsa.s:301: Error: unrecognized symbol type ""
C:\msys64\tmp\ccNIJEsa.s:307: Error: unrecognized symbol type ""
C:\msys64\tmp\ccNIJEsa.s:313: Error: unrecognized symbol type ""
ninja: build stopped: subcommand failed.

carlescufi, 2018-09-13:
You are trying to build for an intel board (Arduino 101) with GNU Arm Embedded, an Arm compiler.

carlescufi, 2018-10-04:
Closing due to lack of feedback and availability of native builds for Windows

closed 2018-10-04 by carlescufi

n3rd4i, 2019-06-27:
is there any new/workaround/updates on this issue ?

saad-abdullah62, 2021-02-15:
@carlescufi Hi, I am trying to use GNU ARM Embedded to build application for qemu_cortex_a53 board and getting the same error. I tried to build application for bcm958402m2_a72 also which is using cortex-a72. But getting same results. I can build application for both of them using aarch64 toolchain in zephyr-sdk but i am trying to add zedboard and zynq7000 that uses cortex A9 so i can't use aarch64 for cortex A9. Any workaround will be helpful.

carlescufi, 2021-02-23:
MSYS is not a supported platform for Zephyr. If you are trying to build on Windows then please use standard `cmd.exe` or WSL (or WSL2) and then report back.