zephyrproject-rtos/zephyr #9960: Stack check test fails - qemu_x86:tests/kernel/fatal/kernel.common.stack_protection

b0661, 2018-09-13:
```
***** Kernel OOPS! *****
Current thread ID = 0x00400060
eax: 0x00000202, ebx: 0x00000000, ecx: 0x80000000, edx: 0x00413060
esi: 0x00000000, edi: 0x0000185e, ebp: 0x00406fcc, esp: 0x00406fc8
eflags: 0x00000002 cs: 0x0008
call trace:
eip: 0x000013a9
     0x0000186e (0x0)
Caught system error -- reason 7
test alt thread 3: initiate kernel panic
***** Kernel Panic! *****
Current thread ID = 0x00400060
eax: 0x00000202, ebx: 0x00000000, ecx: 0x80000000, edx: 0x00413060
esi: 0x00000000, edi: 0x0000185e, ebp: 0x00406fcc, esp: 0x00406fc8
eflags: 0x00000002 cs: 0x0008
call trace:
eip: 0x000013b3
     0x0000186e (0x0)
Caught system error -- reason 8
test stack overflow - timer irq
***** Stack Check Fail! *****
Current thread ID = 0x00400060
eax: 0x0000186e, ebx: 0x00000000, ecx: 0x80000000, edx: 0x00413060
esi: 0x00000000, edi: 0x0000185e, ebp: 0x00406fc4, esp: 0x00404fc4
eflags: 0x00000202 cs: 0x0008
call trace:
eip: 0x00001762
     0x00001796 (0x406fe4)
     0x0000186e (0x0)
Caught system error -- reason 4269956

    Assertion failed at /home/buildslave/src/github.com/zephyrproject-rtos/zephyr/tests/kernel/fatal/src/main.c:233: test_fatal: (crash_reason not equal to _NANO_ERR_STACK_CHK_FAIL)
bad reason code got 4269956 expected 4

FAIL - test_fatal
===================================================================
===================================================================
PROJECT EXECUTION FAILED
```

spoorthik, 2018-12-18:
Same test also fails on nrf52840_pca10056, nrf52_pca10040, nrf51_pca10028 and  quark_se_c1000_devboard.

nashif, 2019-01-16:
can't reproduce

closed 2019-01-16 by nashif