zephyrproject-rtos/zephyr #9962: Migrate sensor drivers to device tree

MaureenHelm, 2018-09-13:
- [x] adt7420
- [ ] adxl362
- [ ] adxl372
- [ ] ak8975
- [ ] amg88xx
- [x] apds9960
- [ ] bma280
- [ ] bmc150_magn
- [ ] bme280
- [ ] bmg160
- [ ] bmi160
- [ ] bmm150
- [x] ccs811
- [ ] dht
- [x] fxas21002
- [x] fxos8700
- [x] hdc1008
- [ ] hmc5883l
- [ ] hp206c
- [x] hts221
- [ ] isl29035
- [ ] lis2dh
- [ ] lis3dh
- [x] lis3mdl
- [x] lps22hb
- [x] lps25hb
- [x] lsm6ds0
- [x] lsm6dsl
- [ ] lsm9ds0_gyro
- [ ] lsm9ds0_mfd
- [x] max30101
- [ ] max44009
- [ ] mcp9808
- [x] mma8451q
- [ ] mpu6050
- [ ] nrf5
- [ ] pms7003
- [ ] sht3xd
- [ ] sx9500
- [ ] th02
- [ ] tmp007
- [ ] tmp112
- [x] vl53l0x

MaureenHelm, 2018-11-27:
> I can take the ST sensors (lis*, lsm*) work.

Thanks!

> (Please note that in the above list the lis2mdl is missing as it has been recently added)

I added it to the list.

avisconti, 2018-12-03:
@MaureenHelm 
Ticked lis2dh in the list above and moved on lis3dh.

avisconti, 2018-12-10:
@MaureenHelm 
lis2mdl is ready but requires your review. (PR #11832)

avisconti, 2019-01-04:
@MaureenHelm 
For completeness I added in the list above also lsm303dlhc accel/magn driver because they wer missing.
The good news is that they already use info from DT!

galak, 2019-11-05:
@Mani-Sadhasivam can you look at converting the ccs811 to DTS?

Mani-Sadhasivam, 2019-11-05:
@galak Okay. Will do it this weekend.

galak, 2019-11-05:
> @galak Okay. Will do it this weekend.

thanks

pabigot, 2019-11-05:
A more complete list of sensors that need to be converted is in #20017; could we close this and use that one?  @galak

pabigot, 2019-11-05:
I've transferred the entries added here to #20017 so we don't have to look in two places to see what's being worked on.  The original list there only focused on uses that included interrupts, so I'll be adding a few more to that issue.

galak, 2019-11-05:
> I've transferred the entries added here to #20017 so we don't have to look in two places to see what's being worked on. The original list there only focused on uses that included interrupts, so I'll be adding a few more to that issue.

My changes are purely DT changes, I didn't make any conversions to new GPIO API.  So not sure about moving this list into #20017, unless we want that PR to handle both changes (DT & updated GPIO API support).

microbuilder, 2019-11-05:
@MaureenHelm I'm digging around to see what sensors I have from this list, but found a `hmc5883l` so I can take that one, and I'll see if I can find any other parts that don't have owners yet.

galak, 2019-11-06:
> @Mani-Sadhasivam can you look at converting the ccs811 to DTS?

Looks like it was pretty easy so I did it #20347

galak, 2020-02-20:
#22963 and #22962 should close this out.

MaureenHelm, 2020-03-10:
> #22963 and #22962 should close this out.

Merged

closed 2020-03-10 by MaureenHelm