zephyrproject-rtos/zephyr #9964: The SDK is are install not  fully, only the compiler libraries are installed

slavaza, 2018-09-13:
The SDK is are install not  fully, only the compiler libraries are installed. But compilers do not installed. Is this how it should be? The installer says success. (Linux Debian 9.0 Netrunner 17)

SebastianBoe, 2018-09-13:
Compilers are included in the SDK.

If you haven't seen an error message yet I would recommend continuing until you see one.

slavaza, 2018-09-13:
This is a second attempt.
master@home-pc:~/Downloads$ bash zephyr-sdk-0.9.3-setup.run
Verifying archive integrity... All good.
Uncompressing SDK for Zephyr  100%  
Enter target directory for SDK (default: /opt/zephyr-sdk/): /home/master/Programs/zephyr-sdk-0.9.3/
Installing SDK to /home/master/Programs/zephyr-sdk-0.9.3
The directory /home/master/Programs/zephyr-sdk-0.9.3/sysroots will be removed! 
Do you want to continue (y/n)? 
y
 [*] Installing x86 tools... 
 [*] Installing arm tools... 
 [*] Installing arc tools... 
 [*] Installing iamcu tools... 
 [*] Installing mips tools... 
 [*] Installing nios2 tools... 
 [*] Installing xtensa tools... 
 [*] Installing riscv32 tools... 
 [*] Installing additional host tools... 
Success installing SDK. SDK is ready to be used.
master@home-pc:~/Downloads$ y
y: command not found
master@home-pc:~/Downloads$ bash zephyr-sdk-0.9.3-setup.run --help
Makeself version 2.2.0
 1) Getting help or info about zephyr-sdk-0.9.3-setup.run :
  zephyr-sdk-0.9.3-setup.run --help   Print this message
  zephyr-sdk-0.9.3-setup.run --info   Print embedded info : title, default target directory, embedded script ...
  zephyr-sdk-0.9.3-setup.run --lsm    Print embedded lsm entry (or no LSM)
  zephyr-sdk-0.9.3-setup.run --list   Print the list of files in the archive
  zephyr-sdk-0.9.3-setup.run --check  Checks integrity of the archive
 
 2) Running zephyr-sdk-0.9.3-setup.run :
  zephyr-sdk-0.9.3-setup.run [options] [--] [additional arguments to embedded script]
  with following options (in that order)
  --confirm             Ask before running embedded script
  --quiet               Do not print anything except error messages
  --noexec              Do not run embedded script
  --keep                Do not erase target directory after running
                        the embedded script
  --noprogress          Do not show the progress during the decompression
  --nox11               Do not spawn an xterm
  --nochown             Do not give the extracted files to the current user
  --target dir          Extract directly to a target directory
                        directory path can be either absolute or relative
  --tar arg1 [arg2 ...] Access the contents of the archive through the tar command
  --                    Following arguments will be passed to the embedded script
master@home-pc:~/Downloads$ bash zephyr-sdk-0.9.3-setup.run --lsm
No LSM.
master@home-pc:~/Downloads$ bash zephyr-sdk-0.9.3-setup.run --list
Target directory: toolchains
drwxr-xr-x root/root         0 2018-05-12 02:37 ./
-rwxr-xr-x root/root  32498254 2018-05-12 02:13 ./zephyr-sdk-baremetal-x86_64-meta-toolchain-iamcu-toolchain-2.2.sh
-rwxr-xr-x root/root  25974838 2018-05-12 02:11 ./zephyr-sdk-baremetal-x86_64-meta-toolchain-xtensa-toolchain-2.2.sh
-rwxr-xr-x root/root  32290308 2018-05-12 02:14 ./zephyr-sdk-baremetal-x86_64-meta-toolchain-mips32r2-toolchain-2.2.sh
-rwxr-xr-x root/root      5672 2018-05-12 02:34 ./make_zephyr_sdk.sh
-rw-r--r-- root/root      3990 2018-05-12 02:34 ./template_dir
-rwxr-xr-x root/root  26772599 2018-05-12 02:12 ./zephyr-sdk-baremetal-x86_64-meta-toolchain-riscv32-toolchain-2.2.sh
-rwxr-xr-x root/root  32946748 2018-05-12 02:12 ./zephyr-sdk-baremetal-x86_64-meta-toolchain-i586-toolchain-2.2.sh
-rwxr-xr-x root/root  42421438 2018-05-12 02:21 ./zephyr-sdk-baremetal-x86_64-meta-toolchain-armv5-toolchain-2.2.sh
-rwxr-xr-x root/root  47365463 2018-05-12 02:17 ./zephyr-sdk-baremetal-x86_64-meta-toolchain-arc-toolchain-2.2.sh
-rwxrwxrwx root/root      6255 2018-05-12 02:37 ./setup.sh
-rwxr-xr-x root/root  30786841 2018-05-12 02:13 ./zephyr-sdk-baremetal-x86_64-meta-toolchain-nios2-toolchain-2.2.sh
-rwxr-xr-x root/root  35380254 2018-05-12 02:34 ./zephyr-sdk-x86_64-hosttools-standalone-0.9.sh

SebastianBoe, 2018-09-13:
Right, and now run

find /home/master/Programs/zephyr-sdk-0.9.3/

slavaza, 2018-09-13:
I found by printing the tree. This is unusual place. Excuse me. Thank you. Excuse me again.

closed 2018-09-13 by slavaza

SebastianBoe, 2018-09-13:
No worries.

slavaza, 2018-09-13:
Ok. I would put two stars.