zephyrproject-rtos/zephyr #9972: Porting to a new architecture needs to be documented

pfl, 2018-09-13:
Comprehensive information about which functions and symbols are expected by the Zephyr kernel when porting to a new architecture is missing. This not only applies to a new port, but also to enhancing an existing port, such as the native posix one.

Porting is split over /arch and /board, with the documentation indicating too vaguely what functionality is expected where, and also which features/functions are expected and by which core subsystem. The different subsystems are mentioned on the top level porting page, but not exactly which functions/symbols are expected by the port. Trial and error is fine, but leads quickly to cargo cult programming should one copy-paste a very similar port for another board.

pfl, 2018-09-13:
Not having a board to port but instead zooming in on the pseudo posix port from a Zephyr application starting point makes me want to have a bit more details on the symbols/functions that are needed by a Zephyr port. I can detail the documentation needed, but only after I have ported/improved something, after which I would not need isuch documentaton either...

nashif, 2019-01-30:
Going to close this, we do have a porting guide already.

closed 2019-01-30 by nashif