zephyrproject-rtos/zephyr #9978: soc: stm32: Extra tree soc configuration is blocked by /dts/arm/st/mem.h

ibisz, 2018-09-14:
After configuring an extra tree SOC based on documentation from #9776 "Move soc code under ZEPHYR_PROJECT/soc" , the build stopps with:
```
.../mem.h:128:2: error: #error "Flash, RAM, and CCM sizes not defined for this chip"
 #error "Flash, RAM, and CCM sizes not defined for this chip"
  ^~~~~
```

erwango, 2018-09-14:
To better understand your issue, are you using a STM32 out of tree?
Or is that a STM32 point blocking other SoCs as well?

erwango, 2018-09-14:
@ibisz , would the following work?
 
```
-#error "Flash, RAM, and CCM sizes not defined for this chip"
+#ifndef DT_FLASH_SIZE
+#ifndef DT_SRAM_SIZE
+#error "Flash, RAM, or CCM sizes not defined for this chip"
+#endif /* DT_SRAM_SIZE */
+#endif /* DT_FLASH_SIZE */
```

erwango, 2018-09-14:
@ibisz : Ok, thanks, will push it.
Btw, I'm interested by your feedback of using STM32 SoC out of tree.
Don't hesitate to raise other issues, even for enhancements!

PR #9980 (2018-09-15): dts/st: enables possibility to define Flash and SRAM out of tree
In case of out of tree SoC, mem.h will trigger and error since
it is not aware of new SoC CONFIG_SOC_<reference>.
Modify error condition in order to generate the error only if
the flash and error size are not defined at this point (even in
out of tree case).

Fixes #9978
commit b6806883 (2018-09-15): dts/st: enables possibility to define Flash and SRAM out of tree

In case of out of tree SoC, mem.h will trigger and error since
it is not aware of new SoC CONFIG_SOC_<reference>.
Modify error condition in order to generate the error only if
the flash and error size are not defined at this point (even in
out of tree case).

Fixes #9978
closed 2018-09-15 by nashif