zephyrproject-rtos/zephyr #9992: cmake compiler cache failures in CI

nashif, 2018-09-14:
see https://app.shippable.com/github/zephyrproject-rtos/zephyr/runs/22294/4/console

```
-- Performing Test toolchain_is_ok - Success
CMake Error at /home/buildslave/.cache/zephyr/ToolchainCapabilityDatabase.cmake:76:
  Parse error.  Function missing ending ")".  End of file reached.
Call Stack (most recent call first):
  ../../../cmake/extensions.cmake:638 (include)
  ../../../cmake/extensions.cmake:1020 (zephyr_check_compiler_flag)
  ../../../cmake/extensions.cmake:996 (target_cc_option_fallback)
  ../../../cmake/extensions.cmake:106 (target_cc_option)
  ../../../CMakeLists.txt:212 (zephyr_cc_option)

```

SebastianBoe, 2018-09-17:
Will get this fixed. Any information / half-baked-theories would be useful ...

SebastianBoe, 2018-09-17:
Possible causes:

- [ ] The process that writes to the cache is killed mid-line
- [ ] Some cache entries are not parsable as a set(a b) code.
- [ ] The process that reads the cache does so while another process is mid-line in a write

nashif, 2018-09-20:
https://app.shippable.com/github/zephyrproject-rtos/zephyr/runs/22778/5/console

SebastianBoe, 2018-09-21:
I see from those logs that there is a new symptom (or a new root cause). CMake runs the command

```
  file(
    APPEND
    ${ZEPHYR_TOOLCHAIN_CAPABILITY_CACHE}
    "set(${key} ${inner_check})\n"
    )

```
and fails with the error:

```
CMake Error at ../../../../cmake/extensions.cmake:671 (file):
  file failed to open for writing (No such file or directory):

    /home/buildslave/.cache/zephyr/ToolchainCapabilityDatabase.cmake
Call Stack (most recent call first):
  ../../../../cmake/extensions.cmake:1020 (zephyr_check_compiler_flag)
  ../../../../cmake/extensions.cmake:996 (target_cc_option_fallback)
  ../../../../cmake/extensions.cmake:106 (target_cc_option)
  ../../../../CMakeLists.txt:210 (zephyr_cc_option)

```

I cannot yet explain how this can happen. But I can say that it is not a permission issue, as I have tested what happens when you have insufficient permissions, and the error message is different (Permission denied). Note that file(APPEND should create the file when it doesn't exist, so there is no user-error here.

Will continue investigating ...

SebastianBoe, 2018-09-24:
Have added more debug information in PR #11027 . Will investigate when it triggers with the same symptoms again (AFAICT it has not triggered for a week now).

PR #10481 (2018-10-19): cmake: Re-write the capability database to avoid file appends
CI sometimes fails with a temporarily corrupted toolchain capabilitiy
database file. Although not proven, there is evidence that CMake's
file(APPEND does not work atomically when there are concurrent writes
and reads of a certain size.

To avoid file appending, we re-write the key-value database
implementation to store keys in filenames and values in individual
single-byte files.

This (most likely) fixes #9992.

NB: Users that have been overriding the database file location with
the CMake variable ZEPHYR_TOOLCHAIN_CAPABILITY_CACHE  [... cut]
commit e7dcc9de (2018-10-19): cmake: Re-write the capability database to avoid file appends

CI sometimes fails with a temporarily corrupted toolchain capabilitiy
database file. Although not proven, there is evidence that CMake's
file(APPEND does not work atomically when there are concurrent writes
and reads of a certain size.

To avoid file appending, we re-write the key-value database
implementation to store keys in filenames and values in individual
single-byte files.

This is (most likely) fixes #9992.

NB: Users that have been overriding the database file location with
the CMake variable ZEPHYR_TOOLCHAIN_CAPABILITY_CA [... cut]
closed 2018-10-19 by nashif