zephyrproject-rtos/zephyr #9993: Few error codes in POSIX API implementation is not supported

spoorthik, 2018-09-15:
I referred [IEEE Std 1003.1-2017](http://pubs.opengroup.org/onlinepubs/9699919799/) documentation and figured out that some of the POSIX API implementation doesn't support all the return values and error codes. Are these listed error codes are optional to support? If so have we documented it somewhere? 

POSIX API | Inconsistency observed in Zephyr implementation
-- | --
pthread_create | Return values,   EAGAIN and EPERM are not supported.
pthread_join | Return value -   EDEADLK not supported. pthread_join(self) should return EDEADLK.
pthread_getschedparam | Return value -   only success scenario return code is implemented. Error codes    are not   implemented.
pthread_attr_setstack | Return value -   [EINVAL] is not supported.
pthread_attr_setschedpolicy | Return value   ENOTSUP is not supported.
pthread_attr_getschedparam | Return value error   code ENOTSUP is not supported.
clock_gettime | EOVERFLOW is not   supported.
clock_settime | EPERM is not   supported.
pthread_rwlock_init | EAGAIN, ENOMEM,   EPERM are not supported.
pthread_rwlock_rdlock | EDEADLK is not   supported.
pthread_rwlock_timedrdlock | EAGAIN and EDEADLK   are not supported.
pthread_rwlock_wrlock | EDEADLK is not   supported.
pthread_key_create | EAGAIN is not   supported.
sem_getval | EINVAL is not   supported.
sem_init | ENOSPC and EPERM   are not supported.
sem_timedwait | EDEADLK and EINTR   are not supported.
sem_trywait | EDEADLK and EINTR   are not supported.
timer_create | EAGAIN and ENOTSUP   are not supported.

spoorthik, 2018-09-17:
FYI: @ramakrishnapallala @nashif

punitvara, 2018-09-24:
@spoorthik @ManojSubbarao This is not bug. To support each error code, unsupported or todo may need to added or rearrange some of the code.

spoorthik, 2018-09-26:
@punitvara @ramakrishnapallala: Shouldn't we document the error codes which are unsupported somewhere?

nniranjhana, 2018-09-27:
@spoorthik is right. Do we have documentation for this anywhere? If not, we need to have it, and also explanation for why they are unsupported. What do you think @nashif @andyross 

I have a patch fixing few error codes at #10173 

Here's my two cents on the _pthread.c_ functions (from the table above):

- `pthread_create`
    I've separated the EINVAL check and added EAGAIN for limit on number of threads.
EINVAL, according to doc is only for
    > Invalid settings in attr

    Now, we're only missing `EPERM`, which is for
    > No permission to set the scheduling policy and parameters specified in attr.

    AFAIK, this error code doesn't make sense for us, since pthreads are created as kernel threads only, and so there's no question of permissions unless we plan to support USERSPACE in our POSIX implementation. Is there any plan for that?

- `pthread_join`
    In the newer revision (Issue 7), EDEADLK seems to be deprecated with the following reason
    > The [EDEADLK] error for the calling thread is removed; this condition results in undefined behavior.

    Do we still need to support this?

- `pthread_getschedparam`
    I've added the ESRCH error code return. The other error codes are valid only for setschedparam.

- `pthread_attr_setstack`
    Okay, so here the EINVAL is when either the stack is greater or lesser than a limit, and when alignment is not met. Both of these are done extensively in the `k_thread_create` call in our kernel, but there's no explicit return for them as we return only the thread ID after creation. And so this error code was not added in POSIX, I think. Do we still need to add a superficial check here, just for supporting the error code?

- `pthread_attr_setschedpolicy`
    ENOTSUP is when
    > An attempt was made to set the attribute to an unsupported value.

    I don't get this, since when we do the EINVAL check, we include for the case when
    > The priority specified in param does not make sense for the current scheduling policy of attr.

    Do we detect an attempt to set for this error? Well, doc also says this error is optional (and unrelated, Linux doesn't support it).

    Anyway, I have removed this error return from `pthread_attr_setschedparam`, since EINVAL is the right error for the policy/priority invalid check and ENOTSUP is an operation error.

- `pthread_getschedparam`
    ENOTSUP is only for setschedparam and setschedpolicy; not for this API.

pfalcon, 2018-10-15:
I don't know why this issue is at all exists.

> Are these listed error codes are optional to support?

Of course, the spec just lists typical error codes which can happen for each call. That doesn't mean that they all should be used in a particular implementation. Nor it even means that an implementation can't return some other error. (However, it means that if an implementation faces and handles some error condition described in the spec, then the error code described by the spec should be returned, not something else).

punitvara, 2018-10-22:
yes I agree @pfalcon  However I will try to add where it is possible to add error code if functionality exist at the moment

punitvara, 2018-10-29:
sem_init - We are already checking whether we are trying to init more than max value or not. Then I dont know what ENOSPC makes sense as it is also checking if semaphore limit. It is not possible to whether any resource exhausted or not because there are no function with return value. As @nniranjhana said we are spawning kernel thread so EPREM also doesn't make sense as of now.

sem_timedwait, sem_trywait - I think we can't check because in semaphore we don't have ownership concept as mutex. EDEADLK can't be supported please support if there is otherway to return this @pfalcon .

PR #10606 (2018-10-31): lib: posix: Return errno code
Return EINVAL errno when argument doesn't refer to
valid semaphore.

partly fixes #9993
commit 2cf27075 (2018-10-31): lib: posix: Return errno code

Return EINVAL errno when argument doesn't refer to
valid semaphore.

partly fixes #9993
closed 2018-10-31 by nashif