zephyrproject-rtos/zephyr #10022: Porting Modbus Library: Need some guidance

Josh0027, 2018-09-17:
Good day! I am just starting to be active on this project and after my last botched PR, I need some advice and direction.
I am attempting to port this ModBus Library to Zephyr and although I am still learning on the steps, I can't seem to successfully compile it or link it. This is the link to the original library: 

https://github.com/Jacajack/liblightmodbus

I first built the library by itself and after verification I proceeded to look for a template code.

I have found and build the /samples/application_development/external_lib and flashed it to my Nucleo board to verify it. Afterwards I used this as my base and cloned the Modbus Library.

First problem is that the external_lib template uses make but the Modbus Library uses cmake first before make. 

Second problem is after I disabled the "ExternalProject_Add" and proceeded in linking the prebuilt library. I suddenly got "liblightmodbus/build/liblightmodbus.a: error adding symbols: File format not recognized".

I'm still in the process of learning but right now I need any help on where I can identify where my mistake is. Any advice is appreciated.

Here is my most recent log when I just attempted to link the prebuilt library.

`josh@ubuntu1604vm:~/SambaShared/zephyr/samples/application_development/ZModbus/build/nucleo_l476rg$ cmake -GNinja -DBOARD=nucleo_l476rg ../..
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.5.2", minimum required is "3.4") 
-- Selected BOARD nucleo_l476rg
Zephyr version: 1.13.0
Parsing Kconfig tree in /home/josh/SambaShared/zephyr/Kconfig
Loading /home/josh/SambaShared/zephyr/boards/arm/nucleo_l476rg/nucleo_l476rg_defconfig as base
Merging /home/josh/SambaShared/zephyr/samples/application_development/ZModbus/prj.conf
-- Generating zephyr/include/generated/generated_dts_board.h
nucleo_l476rg.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller@48000000/usart1@0 has a unit name, but no reg property
nucleo_l476rg.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller@48000000/usart1@1 has a unit name, but no reg property
nucleo_l476rg.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller@48000000/usart1@2 has a unit name, but no reg property
nucleo_l476rg.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller@48000000/usart2@0 has a unit name, but no reg property
nucleo_l476rg.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller@48000000/usart2@1 has a unit name, but no reg property
nucleo_l476rg.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller@48000000/usart2@2 has a unit name, but no reg property
nucleo_l476rg.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller@48000000/usart2@3 has a unit name, but no reg property
nucleo_l476rg.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller@48000000/usart3@0 has a unit name, but no reg property
nucleo_l476rg.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller@48000000/can@0 has a unit name, but no reg property
nucleo_l476rg.dts_compiled: Warning (unit_address_vs_reg): Node /leds/led@0 has a unit name, but no reg property
nucleo_l476rg.dts_compiled: Warning (unit_address_vs_reg): Node /gpio_keys/button@0 has a unit name, but no reg property
-- Cache files will be written to: /home/josh/.cache/zephyr
-- The C compiler identification is GNU 6.2.0
-- The CXX compiler identification is GNU 6.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/arm-zephyr-eabi/arm-zephyr-eabi-gcc
-- Performing Test toolchain_is_ok
-- Performing Test toolchain_is_ok - Success
-- 
*******************************
*** NOTE TO KCONFIG AUTHORS ***
*******************************

The behavior of Kconfig 'default' properties in Zephyr has changed. The 
earliest default with a satisfied condition is now used, instead of the 
last one. This is standard Kconfig behavior.

See http://docs.zephyrproject.org/porting/board_porting.html#old-zephyr-kconfig-behavior-for-defaults.

To get rid of this note, create a file called 'hide-defaults-note' in the 
Zephyr root directory. An empty file is fine.
-- Configuring done
-- Generating done
-- Build files have been written to: /home/josh/SambaShared/zephyr/samples/application_development/ZModbus/build/nucleo_l476rg
josh@ubuntu1604vm:~/SambaShared/zephyr/samples/application_development/ZModbus/build/nucleo_l476rg$ sudo ninja
[4/115] Preparing syscall dependency handling

[110/115] Linking C executable zephyr/zephyr_prebuilt.elf
FAILED: : && ccache /opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/arm-zephyr-eabi/arm-zephyr-eabi-gcc    zephyr/CMakeFiles/zephyr_prebuilt.dir/misc/empty_file.c.obj  -o zephyr/zephyr_prebuilt.elf  -T zephyr/linker.cmd -Wl,-Map=/home/josh/SambaShared/zephyr/samples/application_development/ZModbus/build/nucleo_l476rg/zephyr/zephyr.map -u_OffsetAbsSyms -u_ConfigAbsSyms -Wl,--whole-archive libapp.a zephyr/libzephyr.a zephyr/arch/arm/core/libarch__arm__core.a zephyr/arch/arm/core/cortex_m/libarch__arm__core__cortex_m.a zephyr/arch/arm/core/cortex_m/mpu/libarch__arm__core__cortex_m__mpu.a zephyr/lib/libc/minimal/liblib__libc__minimal.a zephyr/boards/boards/arm/nucleo_l476rg/libboards__arm__nucleo_l476rg.a zephyr/drivers/gpio/libdrivers__gpio.a zephyr/drivers/pwm/libdrivers__pwm.a zephyr/drivers/serial/libdrivers__serial.a -Wl,--no-whole-archive zephyr/kernel/libkernel.a zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj -L"/opt/zephyr-sdk/sysroots/armv5-zephyr-eabi/usr/lib/arm-zephyr-eabi/6.2.0/armv7e-m" -L/home/josh/SambaShared/zephyr/samples/application_development/ZModbus/build/nucleo_l476rg/zephyr -lgcc -Wl,--print-memory-usage ../../liblightmodbus/build/liblightmodbus.a -nostartfiles -nodefaultlibs -nostdlib -static -no-pie -Wl,-X -Wl,-N -Wl,--gc-sections -Wl,--build-id=none && :
../../liblightmodbus/build/liblightmodbus.a: error adding symbols: File format not recognized
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
josh@ubuntu1604vm:~/SambaShared/zephyr/samples/application_development/ZModbus/build/nucleo_l476rg$ 
`

Josh0027, 2018-09-17:
Here is the updated log after I ported the library into Zephyr. Please advise me where I have made the mistake. 

[stdout.log](https://github.com/zephyrproject-rtos/zephyr/files/2388720/stdout.log)

[CMakeLists.txt](https://github.com/zephyrproject-rtos/zephyr/files/2388723/CMakeLists.txt)

Josh0027, 2018-09-17:
Second scenario is I force the cmake to generate Unix Makefiles but it gives ma an error.  It looks like mixing -GNInja and -GUnix Makefiles is not good. Maybe this is also the reason I am seeing my first issue above?

Any ideas are welcome.

[stdout2.log](https://github.com/zephyrproject-rtos/zephyr/files/2389600/stdout2.log)

[CMakeLists2.txt](https://github.com/zephyrproject-rtos/zephyr/files/2389601/CMakeLists2.txt)

firmwareguru, 2019-11-07:
@Josh0027 Did you get anywhere with this, and where is the repository you've been working in?

parthitce, 2020-08-08:
@firmwareguru @Josh0027 Although we have modbus RTU on it's way for in-tree Zephyr (https://github.com/zephyrproject-rtos/zephyr/pull/26516), I did tried to use liblightmodbus for my personal use case. Adding my attempts just for the completion of this thread.

Am able to successfully compile and use the sample in zephyr.

Changes in liblightmodbus is here: https://github.com/linumiz/liblightmodbus
Changes with example in zephyr is here: https://github.com/linumiz/zephyr/tree/liblightmodbus

closed 2020-10-22 by github-actions[bot]