zephyrproject-rtos/zephyr #10037: fade_led sample doesn't work with pwm_nrfx driver

JoeHut, 2018-09-17:
I tried to change the fade_led sample to support the pwm_nrfx driver introduced in v1.13. This results in a fail to set the pwm period.
Reason: The default in the pwm_nrfx driver for prescalers is 0, which means that a 16MHz clock is applied. The sample therefore sets a period of 320,000 cycles but the driver only supports max 32,768 cycles, so the clock has to be reduced to a frequency of 1MHz to make the sample work. This seems to be odd since the software based pwm driver supports 50us as period with a 16MHz clock.

My question basically is just, whether this is intended and the prescaler should be adjusted if support for the pwm_nrfx driver is added to the samples or if there is something wrong.

nashif, 2019-01-30:
@carlescufi can you please assign?

carlescufi, 2019-01-30:
Added @pdunaj as well since we have similar functionality downstream

anangl, 2019-02-01:
This 32,767 cycles limit is no longer present in the pwm_nrfx driver. Since commit 2d5ea10f314a12b74269ad5d88408445057a8b6d, the PWM prescaler is modified accordingly to find a fit for the requested period length, precisely [this part of code](https://github.com/zephyrproject-rtos/zephyr/blob/2d5ea10f314a12b74269ad5d88408445057a8b6d/drivers/pwm/pwm_nrfx.c#L56) does this.
I think this issue can be closed as resolved by #11726.

anangl, 2019-02-12:
Fixed by #11726.

closed 2019-02-12 by anangl