zephyrproject-rtos/zephyr #10041: MISRA C - types that indicate size and signedness should be used instead of basic numerical types

ceolin, 2018-09-17:
The basic numerical types of char, short, int, long, long long (C99), float, double and long double should not be used, but specific-length typedefs should be used.

In Zephyr we have, `zephyr/include/types.h` defining some specific-length types and we also have `stdint.h` definitions. On top of that several APIs are using neither of them. We must define one of them to use.

*note* this is not required when interfacing with the standard library or with code outside the project's control.

part of #9552

ceolin, 2018-09-26:
dir 4.6

d3zd3z, 2019-12-20:
Should we revisit the decision in Zephyr to define our own integer types? Most of the code I work on these days isn't exclusive to Zephyr, and therefore has to use `stdint.h` types, instead of Zephyr-specific types. There is quite of bit of this outside of the core code.

Fortunately, C is loose with types, and mixing between `u8_t` and `uint8_t` doesn't generally hurt anything (other than print formatting). So, maybe we don't need to mandate one of these in particular.

closed 2021-05-28 by nashif