zephyrproject-rtos/zephyr #10043: RISCv32 qemu configuration does not work with upstream qemu

nashif, 2018-09-18:
Tried latest qemu and no luck running Zephyr applications with the latest version from master.

@nategraff-sifive any hints?

nategraff-sifive, 2018-09-18:
Latest QEMU has different machines than the QEMU in Zephyr SDK. You'll want to build with "-DBOARD=hifive1" and then run QEMU with the argument "-machine sifive_e".

We should point the qemu_riscv32 board at the riscv32_fe310.dtsi since the "sifive_e" target is equivalent to an SiFive FE310, the same as on the HiFive1.

nashif, 2018-09-21:
@nategraff-sifive thanks, that works, I am getting a few issues on riscv32 now though, not sure what is going on, can you reproduce those on the actual hifive1 board?

See https://app.shippable.com/github/zephyrproject-rtos/zephyr/runs/22802/3/tests

nashif, 2018-09-21:
@nategraff-sifive note that the above is using a pre-release of the SDK, so you might want to download that version of the SDK to reproduce or using a recent qemu.

https://github.com/zephyrproject-rtos/meta-zephyr-sdk/releases/tag/0.9.4-rc3

nategraff-sifive, 2018-09-21:
@nashif I'll take a look 👍

nashif, 2018-10-09:
@nategraff-sifive any update?

nategraff-sifive, 2018-10-09:
@nashif Sorry, not yet. I had some other tasks take priority. I should have some time to dive into this within the next few days.

nategraff-sifive, 2018-10-10:
@nashif Are any RISC-V targets part of the standard sanitycheck that gets run on PRs? I went back to establish a baseline and ran `./scripts/sanitycheck -p qemu_riscv32` using SDK 0.9.3 on the `v1.12.0` and `v1.13.0` tags and I saw failures in both cases in:
- tests/kernel/timer/timer_api/kernel.timer
- tests/kernel/timer/timer_monotonic/kernel.timer

Those are two of the tests that are failing with SDK 0.9.4-rc3, so there's also a subset of the regressions that have just been broken for a while.

nashif, 2018-10-11:
qemu_riscv32 is a default platform and all tests run on this board for every commit. I guess what you are seeing is a sporadic behaviour with current qemu that stops being sporadic in the new qemu. usually if you re-run the tests again the test will pass (with sanitycheck --only-failed)

nategraff-sifive, 2018-10-11:
Quick update: I just tested the new_qemu branch against the downstream riscv-qemu and a number of tests are still failing. Notably, with downstream QEMU the tests are crashing QEMU, as opposed to the tests assertions failing in upstream. It remains to be seen which of the two failures is preferable.

nategraff-sifive, 2018-10-12:
@nashif

So the QEMU sifive_e target doesn't quite *exactly* match the FE310-G000 SoC on the HiFive 1. One of the discrepancies is that the machine timer driving Zephyr's sys_clock runs at 32768 Hz on the FE310, while in QEMU it runs at the same speed as the CPU, 10 MHz. The result is that tests fail all over the place because they depend on the system clock to be running at the right frequency.

If you set SYS_CLOCK_HW_CYCLES_PER_SEC=10000000 for the HiFive 1, then sanitycheck passes.

This doesn't resolve the actual hardware failure for the `tests/kernel/workq/work_queue` test.

I think the proper resolution is the following steps:
- Bring back the qemu_riscv32 target as a hifive1 clone with this configuration change
- File a bug with QEMU that the machine timer frequency in the sifive_e target differers from the FE310-G000 SoC it's emulating
- Since the work queue bug doesn't affect QEMU or sanitycheck, continue ahead with SDK 0.9.4 and file a bug with the Zephyr project that work queue tests fail on the actual HiFive 1 board so I can track that down without blocking here anymore.

Thoughts?

nategraff-sifive, 2018-10-15:
@galak Doing a quick glance over the source, it looks like modifying `SIFIVE_CLINT_TIMEBASE_FREQ` in [include/hw/riscv/sifive_clint.h](https://github.com/qemu/qemu/blob/master/include/hw/riscv/sifive_clint.h) might just do the trick. Let me try that and see what happens.

nashif, 2018-10-15:
> * Bring back the qemu_riscv32 target as a hifive1 clone with this configuration change

meanwhile I have been testing with the above option and it works fine

nashif, 2018-10-15:
see #10136

nategraff-sifive, 2018-10-15:
@nashif Awesome!

@galak So I made the change I just suggested and ran sanitycheck on the hifive1 target. Oddly enough, there's exactly one test failing and it's `tests/kernel/mem_pool/mem_pool_api/kernel.memory_pool` causing QEMU to jump to 0x00000008, where it subsequently crashes. Not sure yet what's going on, but I'll keep looking.

commit c2c6a6a2 (2018-11-05): qemu_riscv32: use hifive1 configuration

Use hifive1 configuration for this qemu and set
SYS_CLOCK_HW_CYCLES_PER_SEC to 10000000

Fixes #10043
closed 2018-11-05 by nashif