zephyrproject-rtos/zephyr #10054: Logger thread spins forever if there is no backend

aescolar, 2018-09-18:
A configuration with no backend registered will have the logger thread spin forever.

Instead having an assert at the start of the thread would be sensible: https://github.com/zephyrproject-rtos/zephyr/commit/fb0e038c75565b40550bcdf495cd53570a581b0f 

As of now there is 2 Xtensa tests in CI with this issue.

nordic-krch, 2018-10-11:
In case no backends are present in the system we can add assert since in that case it's better to disable logger. What are the issues with xtensa tests? Will it fix it?

aescolar, 2018-10-11:
> In case no backends are present in the system we can add assert since in that case it's better to disable logger.
 
Agreed.
Having an assert would be also a nice way of pointing to the reason of the (configured in) logger not working (there was no backends for it).

> What are the issues with xtensa tests?

That the assert will trigger for them, (assuming they haven't been fixed yet) and they will fail in CI.
Meaning, the assert will work as expected.

> Will it fix it?

An assert does not fix things. It fails noisily when the error condition happens.

nordic-krch, 2018-10-11:
added assert in #10513, if CI fails i'll go from there.

PR #10513 (2019-03-04): logging: Add assert when no backends defined
Added an assert in the logger thread in case there
is no backends, instead of having that thread spinning
forever.

Fixes #10054
closed 2019-03-04 by nashif