zephyrproject-rtos/zephyr #10058: The test mem_pool_threadsafe sporadically hangs forever

SebastianBoe, 2018-09-18:
The test "mem_pool_threadsafe" sporadically hangs forever when run with qemu_cortex_m3.

This causes sporadic CI failures.

It does so about 30% of the time.

Steps to reproduce:

git checkout 8fc5fd55dec1f0808badc5508ca8cca497940f3e
$ZEPHYR_BASE/scripts/sanitycheck -T$ZEPHYR_BASE/tests/kernel/mem_pool/mem_pool_threadsafe -p qemu_cortex_m3

Repeat up to 5 times or so if it does not trigger immediately.

To speed up the testing apply this patch:

```
diff --git a/tests/kernel/mem_pool/mem_pool_threadsafe/testcase.yaml b/tests/kernel/mem_pool/mem_pool_threadsafe/testcase.yaml
index dfa0450076..662b080981 100644
--- a/tests/kernel/mem_pool/mem_pool_threadsafe/testcase.yaml
+++ b/tests/kernel/mem_pool/mem_pool_threadsafe/testcase.yaml
@@ -1,3 +1,4 @@
 tests:
   kernel.memory_pool:
     tags: kernel mem_pool
+    timeout: 10 # Observed to run in 5 sec

```

SebastianBoe, 2018-09-18:
AFAICT this is not a recent regression, but something that has been able to happen for a long time, at least before 1.11.0.

So perhaps it has always been sporadically failing.

andrewboie, 2019-01-17:
I was able to reproduce this in 8fc5fd5 but not in current master, I think some recent fixes @andyross made to mem pools fixes this as well. Closing.

closed 2019-01-17 by andrewboie