zephyrproject-rtos/zephyr #10060: Create a function in BT stack to check if given device is already bonded

rakons, 2018-09-18:
We are missing an API function that would allow us to check if selected device is already bonded.
Usage:
When searching for a device as a HOST I may wish to connect only with the devices that are already bonded. Then I may use passive scanning only and if a device is detected I could have a possibility to check if it was bonded already. Something like:

```
bt_is_bonded(u8_t id, const bt_addr_le_t *addr);
```

jhedberg, 2018-09-18:
Removed the maintainers (@sjanc @Vudentz) from Assignees - I just wanted them to be involved in the discussion

carlescufi, 2018-09-24:
@rakons is this fixed by #10128 ?

rakons, 2018-09-24:
Taking bt_le_set_auto_conn into account, I think we can get rid of this function.

carlescufi, 2018-09-24:
Closing as per [this comment](https://github.com/zephyrproject-rtos/zephyr/issues/10060#issuecomment-423937197)

closed 2018-09-24 by carlescufi