zephyrproject-rtos/zephyr #10105: Bluetooth: GATT find info response

kapi-no, 2018-09-19:
There is probably a bug in ```gatt_find_info_rsp``` function located in the ```subsys/bluetooth/host/gatt.c``` file. Similar GATT response functions may also be affected.

It is assumed in this function that the group of discovered descriptors uses the same type of UUID: either 16-bit or 128-bit one. Now, imagine following scenario:

There is 128-bit charactersteristic with 2 descriptors within it: first one with characteristic value and second one with CCC. In this case, it would be assumed that CCC is 128-bit uuid and would be wrongly parsed (resulting in the callback with ```NULL``` GATT attribute).

hakonfam, 2018-09-20:
_Per response_ it is only valid to have either 16 bit or 128 bit. As specified in the Core spec 5.0. 

[image]

Vudentz, 2018-09-20:
Nope, it actually checks if attribute is of the same type an otherwise it stops:

https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/bluetooth/host/att.c#L437

I know it works since BlueZ does use it and I got correct responses.

kapi-no, 2018-09-20:
Yes, you are both right. I noticed a problem with GATT discovery in the aforementioned scenario, but I incorrectly identified the cause of it. 

Actually, it seems that ```gatt_discover_next``` function (called after splitting the discovery in 2 steps: one for 16-bit descriptor and one for 128-bit descriptor) incorrectly adjusts handle range and doesn't permit to start the discovery procedure with the same start and end handle.

This is addressed in #10119