zephyrproject-rtos/zephyr #10122: API to access bond data

rakons, 2018-09-20:
There currently is no API nor any pre-defined polices what to do with multiple bonds - if new bond is created witch should be removed?

There should be an API to access and identify bonds and delete selected ones. Then some kind of automatic policy configuration may be implemented - but this is nice to have feature. Having an option to access and manually delete bond data is a must.

jhedberg, 2018-09-20:
@Vudentz @sjanc @carlescufi FYI

carlescufi, 2018-09-20:
@rakons what's the rationale behind using an index?

rakons, 2018-09-20:
For your second question, sometimes it would be great to have some user data that may be added to a bond. For example - Central is pairing with keyboard and mouse. When new device is bonding and it is a mouse, I wish to remove old mouse bond. I think that the easiest way to detect that this bond references to a mouse would be if we would have some possible to set and get user data together with a bond.

carlescufi, 2018-09-20:
>sometimes it would be great to have some user data that may be added to a bond.

+1 to that, we have had multiple requests in the past in this direction. That said, this can of course be achieved at the application level, but I think it makes sense to incorporate it in the Bluetooth subsystem.

rakons, 2018-09-20:
@carlescufi RAM usage - you do not have to copy all the data into the memory, you may iterate through them, but just simpler than using foreach construction that uses callback, just:

```
addr;
for(size_t i=0; bt_le_get_bond(&addr, i); ++i)
{
// process
}
```

carlescufi, 2018-09-20:
@rakons right. Not sure if I like the concept of an index, but then maybe have an iterator instead (`start()`, `next()`). Thoughts?

rakons, 2018-09-20:
@carlescufi Only if it would work safely if multiple threads would like to iterate the bonds - maybe it makes not a big sense to iterate bonds from multiple threads, but is should be safe at least.

carlescufi, 2018-09-20:
@rakons good point. Let's hear from @jhedberg and the others as well.

jhedberg, 2018-09-20:
A "foreach" helper would at least be consistent with existing practice, e.g.:

```
void bt_gatt_foreach_attr(u16_t start_handle, u16_t end_handle,                 
                          bt_gatt_attr_func_t func, void *user_data);    
```

jhedberg, 2018-09-20:
Also, it feels a bit weird to expose an internal index to the app since it leaks something about the internal implementation (e.g. makes sense if there's an array internally, but would be inefficient if it's a linked list)

jhedberg, 2018-09-20:
Regarding attaching user data to a bond - not opposed to it. The internal representation of a bond is the bt_keys struct, so it'd need to get `void *data'; size_t data_len;` members, and the stack would then take care of writing/reading to/from flash. And then there'd be associated `bt_bond_get_data()` and `bt_bond_set_data()` helpers.

carlescufi, 2018-09-20:
@jhedberg +1 to the `foreach` proposal. I dislike exposing the index as well.

rakons, 2018-09-20:
@jhedberg Be carreful with saving pointers as it might invalidate. I think it is not the best idea to just serialize whole data, it may be just a pointer to a flash sometimes. It may also be pointer to some structure where some fields have to be saved and the rest may be recreated, or the saved values would not be valid after restart. I rather think about moving the duty of saving and restoring user data to the user. Then there should be some callbacks to support it.

jhedberg, 2018-09-20:
@rakons understood - I still think we might be able to do better than leave the storing up to the application. Instead of adding any pointers to bt_keys the get/set APIs could also just map straight to the settings API, i.e. `settings_set_value()` and `settings_get_value()`, i.e. the RAM storage would be provided by the app.

carlescufi, 2018-09-20:
@rakons have you looked at the settings subsystem and how BLE uses it? unfortunately those choices have already been made in order to preserve filesystem independence and for other reasons.

rakons, 2018-09-20:
I am looking into it right now.

rakons, 2018-09-20:
Well... the key-value strings is not the best option in embedded world. As far as string as a key seems reasonable, string as a value is an overkill (still in embedded world).

But, without touching the API itself, we may add just simple base64 coder/decoder to process binary data blobs effectively.

jhedberg, 2018-09-20:
There is work ongoing to do stream-based reading/writing of settings data, that wouldn't require intermediate buffers in PR #9521.

Regarding the settings APIs I mentioned, it's actually not the get/set that this would map to. We may need the app to provide callbacks for data read from flash since it's only read from there upon settings_load().

PR #10128 (2018-09-24): Bluetooth: Add API to iterate through existing bonds
So far the stack hasn't provided any way for the application to access
the existing bonds. This patch adds such an API.

Fixes #10122
commit 559578c0 (2018-09-24): Bluetooth: Add API to iterate through existing bonds

So far the stack hasn't provided any way for the application to access
the existing bonds. This patch adds such an API.

Fixes #10122
closed 2018-09-24 by jhedberg