zephyrproject-rtos/zephyr #10127: Cannot use new shell with native_posix

jukkar, 2018-09-20:
I am seeing this when compiling new net-shell for native_posix and uart.

```
../zephyr/subsys/shell/shell_uart.c:29:36: error: ‘CONFIG_UART_CONSOLE_ON_DEV_NAME’ undeclared (first use in this function); did you mean ‘CONFIG_ETH_NATIVE_POSIX_DEV_NAME’?
  sh_uart->dev = device_get_binding(CONFIG_UART_CONSOLE_ON_DEV_NAME);
                                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                    CONFIG_ETH_NATIVE_POSIX_DEV_NAME
../zephyr/subsys/shell/shell_uart.c:29:36: note: each undeclared identifier is reported only once for each function it appears in
```

This prevents merging of #9825 because sanitychecker fails.

aescolar, 2018-09-20:
there is no UART in native_posix..

pfalcon, 2018-09-20:
It's the same issue as in https://github.com/zephyrproject-rtos/zephyr/pull/5947 (which was skipped, unresolved).

And of course, the baseline problem is that **shell** should not deal with *uart*, **shell** should deal with **console**, and **console** should deal with **uart**. As in many places in Zephyr, we lack proper layering.

aescolar, 2018-09-21:
I will try to add today or so a UART for native_posix.

aescolar, 2018-09-24:
@nordic-krch , @jarz-nordic : Why is the new shell depending on a CONSOLE config option if it is not using it?

pfalcon, 2018-09-24:
@nordic-krch :

> i would also like to see improvements in the console.

You probably should say which console you mean, as we have many "consoles" in Zephyr. If you mean subsys/console/ , yep, it needs further improvements based on the different usecases. RFCs and patches are welcome. Note that I foresee it going into the POSIX direction re: API.

aescolar, 2018-09-24:
I think we need 2 new issues linked to this one:
* The new shell (`CONSOLE_SHELL`) can only be compiled on boards with `SERIAL` but it does not set the dependency in its Kconfig options
* The new shell uses `CONSOLE` kconfig options, even though it does not use `CONSOLE`.

@jarz-nordic : Do you agree? will you create the issues or prefer me to do it for you?

jarz-nordic, 2018-09-24:
Hi,
In my opinion current UART backend implementation for shell must be changed to get rid off CONSOLE dependencies. It should be looking for SERIAL instead.
CONSOLE could be simply one more shell backend.

jukkar, 2018-09-28:
After updating my branch and trying with the fixes in #10212, I still get the same error so some fixes in shell side are still needed. Only if I set `CONFIG_SERIAL=y`, there is no compile error but shell does not work in that case. I have been using `samples/net/echo_server` to test this. This issue is a blocker for net-shell rework at #9825

aescolar, 2018-09-28:
@jukkar can you push that merged branch to your GH repo so I can take a look?

> Only if I set CONFIG_SERIAL=y, there is no compile error

The shell does depend on serial, put as of now it only warns you to compile with it (#10190 3e41ac5f04c88e5ca2443232502368c861609231) . It will not force that dependency. So you do need to set CONFIG_SERIAL.

aescolar, 2018-09-28:
I just merged your branch as you had it in the PR with master, and it seems to compile (after setting CONFIG_SERIAL=y in the project) and run:
[image]

jukkar, 2018-09-28:
Indeed I got it working. I missed the output from native_posix about the pseudo tty as was expecting this to work the same way as when connecting to normal device. Also as networking with native_posix requires root permission, the created pts is also owned by root which means that the screen command must be started with sudo.

Anyway, the console works now but this configuration needs to be seamless. The CONFIG_SERIAL dependency needs to be set somewhere as it is not really practical that is is set in application prj.conf file.

aescolar, 2018-09-28:
> The CONFIG_SERIAL dependency needs to be set somewhere as it is..

You may want to reopen #10190 then. An ugly solution is for SHELL to ``select SERIAL``, another ugly solution is to have a ``default y if SHELL``  in the ``SERIAL`` definition.

jukkar, 2018-09-28:
Ok, lets do that then, closing this one.

closed 2018-09-28 by jukkar