zephyrproject-rtos/zephyr #10137: frdm boards buttons probably don't work

galak, 2018-09-21:
In sample app samples/basic/button/src/main.c

we have:

```
#ifdef SW0_GPIO_FLAGS
#define EDGE    SW0_GPIO_FLAGS
#else
/*
 * If SW0_GPIO_FLAGS not defined used default EDGE value.
 * Change this to use a different interrupt trigger
 */
#define EDGE    (GPIO_INT_EDGE | GPIO_INT_ACTIVE_LOW)
#endif
```

Before we used DTS for buttons, the board.h wasn't setting SW0_GPIO_FLAGS so we'd get the default behavior.  Now that DTS is generating SW0_GPIO_FLAGS, we aren't getting the same flags set.  As most (if not all) have the flag portion set to GPIO_INT_ACTIVE_LOW only.

DWagenk, 2018-09-21:
Just ran into the same problem with stm32 boards (using them for the first time since gpio configuration was moved over to device-tree). Setting the appropiate flags in Boards .dts file solved it. 
```dts
	gpio_keys {
		compatible = "gpio-keys";
		user_button: button@0 {
			label = "User";
			gpios = <&gpioc 13 (GPIO_INT_EDGE | GPIO_INT_ACTIVE_LOW)>;
		};
	};
```
But I guess `GPIO_INT_EDGE` should go to the application, so to button sample in this case, since the decision, if `GPIO_INT_EDGE`, `GPIO_INT_DOUBLE_EDGE` or `GPIO_INT_LEVEL` is used is application-specific and not board-specific. IMO

erwango, 2018-09-21:
Does it happened with  https://github.com/zephyrproject-rtos/zephyr/pull/9551?

dleach02, 2018-09-24:
@DWagenk, to me this seems like a board level configuration since the board has been put together a certain way (unless it is some port/gpio that is more flexible).

DWagenk, 2018-09-24:
@erwango I'm using 1.13.0 with nucleo_f103rb board as my devolpment base right now. Reverting 53cd7c249627c4bc76c15b7d05e330e880a4a977 (from #9551) doesn't have any effect. Reverting 9255bf5f7fc21eb5e6e1ebf4572064a96e5ce9c7 (from #7493) leads to the button sample working out of the box again. 1.12.0 also works out of the box.
9255bf5f7fc21eb5e6e1ebf4572064a96e5ce9c7 now uses `SW0_GPIO_FLAGS` for the buttons interrupt configuration, which is defined for my and as it seems all stm32 boards. Beforehand `SW0_GPIO_INT_CONF` was used, wich was undefined for stm32 boards and led to the default configuration `(GPIO_INT_EDGE | GPIO_INT_ACTIVE_LOW)` beeing used, which is appropiate, at least for the nucleo boards.
So basically the problem is, that neither the button-sample-app nor the stm32 boards dts files provide the necessary `GPIO_INT_EDGE` flag.

@dleach02 I see the `GPIO_INT_ACTIVE_LOW` and `GPIO_PUD_PULL_UP` flags as part of the board level configuration, because it correlates to how I design the hardware. But I think it should be up to the software to select which interrupt type to use. Any opinions or links to past discussion  on this?

dleach02, 2018-09-25:
@DWagenk, if we leave this up to the application software will there be any conflicts on these sample applications where it is configuring the button for multiple boards? It seems like it needs to draw this from the board level DTS so that the application can run across multiple boards without any special conditional code in case one board is active high while another is active low. Unless there is some other way to specify some of these settings that I'm not aware of?

DWagenk, 2018-09-25:
short summary of what exists in boards .dts files so far, found by searching for gpio-keys inside boards dir (in zephyr v1.13):
most boards, including stm32 and frdm boards, set the interrupt polarity to `GPIO_INT_ACTIVE_LOW` explicitly, but don't set any other GPIO flags. Some other boards, including nrf52 boards, set the pull-up configuration to `GPIO_PUD_PULL_UP` and no other flags.  

Not setting any flags implies `(GPIO_INT_ACTIVE_LOW  |  GPIO_INT_LEVEL | GPIO_PUD_NORMAL)`.

@dleach02 I agree, that hardware-specific configuration should go to the boards .dts file. I see `GPIO_INT_ACTIVE_X`,  `GPIO_INT_DEBOUNCE`,  `GPIO_PUD_X` and `GPIO_POL_X` as board-specific. On the other hand I would see `GPIO_INT_LEVEL/EDGE` and `GPIO_INT_DOUBLE_EDGE` as application specific to use the same board-configuration for different scenarios:
- get only _button pressed_ events (`GPIO_INT_EDGE`)
- get _button pressed_ and _button released_ events (`(GPIO_INT_EDGE | GPIO_INT_DOUBLE_EDGE)`)
- get repeated events while the button remains low (`GPIO_INT_LEVEL`)

dleach02, 2018-09-25:
Okay. So does that imply that the basic application sample needs to set the INT_EDGE for all usages? I guess that is my general question because this was initially characterized as a bug in the dts but your position is that it is a bug in the application. If we change it in the application, will this cause any problems with other boards (that I don't have)?

The basic button app has conditional code at the top that makes decisions on the presence of the GPIO_FLAGS where if it isn't defined then set the edge to GPIO_INT_EDGE | GPIO_INT_ACTIVE_LOW. The GPIO_FLAGS for the KW41Z are getting defined by the DTS without the INT_EDGE so the sample fails. Does this code need instead need to define edge in the presence of GPIO_FLAGS as:

`#define EDGE    SW0_GPIO_FLAGS|GPIO_INT_EDGE`

erwango, 2018-09-28:
> The basic button app has conditional code at the top that makes decisions on the presence of the GPIO_FLAGS where if it isn't defined then set the edge to GPIO_INT_EDGE | GPIO_INT_ACTIVE_LOW. 

So I think this code is wrong, `SW0_GPIO_FLAGS` being defined does not carry any information, except it is ... defined.  Same with `SW0_GPIO_PIN`, which by the way triggers and error if not defined.
Only useful pice of code is to check if `SW0_GPIO_NAME` or `SW0_GPIO_CONTROLLER` should be used for `PORT` definition.

Agree with @DWagenk  that this is purely application dependent. Application has to specify what behavior it expects. Board default configuration could not match each application expectation.

Since `sample/basic/button` is here as example, it should provide example on how to configure the expected behavior, so always use:
`#define EDGE    (GPIO_INT_EDGE | GPIO_INT_ACTIVE_LOW)`

Then, maybe we should clean up dts files to provide a neutral configuration (`0`) for these nodes, as any other configuration might be misleading.

dleach02, 2018-09-28:
Discussing this internally, we feel that the dts doe need to reflect the state of the board. I think the primary "bug" here is the conditional and what is needed is to just OR in the GPIO_INT_EDGE as an application level setting.

erwango, 2018-10-01:
> Discussing this internally, we feel that the dts doe need to reflect the state of the board. I think the primary "bug" here is the conditional and what is needed is to just OR in the GPIO_INT_EDGE as an application level setting.

Agreed

nashif, 2019-02-19:
still an issue? @galak @dleach02 @MaureenHelm @erwango

nashif, 2019-02-19:
looks like it was fixed in #10296

closed 2019-02-19 by nashif

galak, 2019-02-19:
It's not fully fixed, but this is more of an enhancement and something that should get cleaned up w/the GPIO/pinmux reworking.