zephyrproject-rtos/zephyr #10143: Why does BT_SETTINGS require PRINTK?

hersp, 2018-09-21:
Hi

How come BT_SETTINGS  requires PRINTK to be enabled ?
(https://docs.zephyrproject.org/1.13.0/reference/kconfig/CONFIG_BT_SETTINGS.html)

I would like to enable BT_SETTINGS without having a console, and I cannot see a reason for having this dependency,.

Peter

nashif, 2018-09-21:
I think vprintk is used here, it is not for printing to console.

jhedberg, 2018-09-21:
Yep, if you check `subsys/bluetooth/host/settings.c` you'll see that there are a few `snprintk()` calls there which require PRINTK to be enabled in Kconfig.

jhedberg, 2018-09-21:
That said, do we have a console dependency for CONFIG_PRINTK? If so, that probably needs to fix so that snprintk, etc are still available even though there's no console.

hersp, 2018-09-21:
CONFIG_PRINTK is dependent of CONSOLE so using snprintf instead would probably be better.

jhedberg, 2018-09-21:
I think it'd be more appropriate to fix the printk-family of functions Kconfig, so that you can select the non-console related ones independent from the console. We use printk since it's more light-weight than printf (e.g. less stack consumption)

jhedberg, 2018-09-21:
Another reason to not use snprintf: it would increase the image size for any build that does have a console, since then you'd have both printk and printf in the image.

hersp, 2018-09-21:
That sounds right, then the current usage of snprintf other places should probably also eventually  be cleaned(e.g. subsys/settings/src/settings.c)

thomas-ot, 2018-10-09:
It seems from the discussion above that there is agreement that console dependencies to printk should be removed. Hence, I'm reopening the issue.

jhedberg, 2018-10-18:
> It seems from the discussion above that there is agreement that console dependencies to printk should be removed. Hence, I'm reopening the issue.

@thomas-ot ok, however then this needs to be re-assigned to some console or prink maintainer.

hersp, 2018-11-15:
Currently I am actually  able to enable BT_SETTINGS when I remove the dependency of PRINTK in subsys/bluetooth/host/Kconfig.  (I ran sanitycheck)
Can the dependency be removed as it seems like it is actually not needed?

jhedberg, 2018-11-27:
> Currently I am actually able to enable BT_SETTINGS when I remove the dependency of > PRINTK in subsys/bluetooth/host/Kconfig. (I ran sanitycheck)
> Can the dependency be removed as it seems like it is actually not needed?

That's surprising. I had to try it myself to believe it. BT_SETTINGS depends on snprintk(), which in turn is defined inside misc/printk.c. I thought misc/printk.c depends on CONFIG_PRINTK, but apparently not? Is that intentional? Before merging the patch to remove the dependency from CONFIG_BT_SETTINGS I'd like to have an answer to this question.

jhedberg, 2018-11-27:
I just found the answer to my own question: take a look at `include/misc/printk.h`. When CONFIG_PRINTK is not set then all these functions (`snprintk()` included) get defined to empty static inline functions. So disabling CONFIG_PRINTK will still keep the build working, but it'll generate a broken image wrt BT_SETTINGS.

pdunaj, 2019-01-22:
Will it be fixed?

hersp, 2019-01-22:
Other work is ongoing on the settings system, maybe this discussion should be moved there.
https://github.com/zephyrproject-rtos/zephyr/issues/12331

nashif, 2019-03-03:
PRINTK does not depend on console anymore, so this issue is considered "fixed"

closed 2019-03-03 by nashif