zephyrproject-rtos/zephyr #10150: Getting LLVM work on ARM

galak, 2018-09-21:
This issue is to track activity related to building Zephyr with LLVM for ARM architectures.

nashif, 2019-02-21:
part of #3689

closed 2019-02-21 by nashif

ochipara, 2019-08-06:
Is the LLVM/ARM support supposed to be functional? I am getting the following error when I try to crosscompile using llvm:

Zephyr version: 1.14.99
-- Found PythonInterp: /anaconda3/envs/zephyr/bin/python (found suitable version "3.7.3", minimum required is "3.4") 
-- Selected BOARD nrf52840_pca10056
-- Found west: /anaconda3/envs/zephyr/bin/west (found suitable version "0.6.0", minimum required is "0.6.0")
-- Invalidating toolchain capability cache in /Users/ochipara/Library/Caches/zephyr
-- Loading /Users/ochipara/Working/embedded/zephyrproject/zephyr/boards/arm/nrf52840_pca10056/nrf52840_pca10056.dts as base
-- Overlaying /Users/ochipara/Working/embedded/zephyrproject/zephyr/dts/common/common.dts
Device tree configuration written to /Users/ochipara/Working/embedded/zephyrproject/zephyr/samples/hello_world/build/zephyr/include/generated/generated_dts_board.conf
Parsing Kconfig tree in /Users/ochipara/Working/embedded/zephyrproject/zephyr/Kconfig
Loaded configuration '/Users/ochipara/Working/embedded/zephyrproject/zephyr/boards/arm/nrf52840_pca10056/nrf52840_pca10056_defconfig'
Merged configuration '/Users/ochipara/Working/embedded/zephyrproject/zephyr/samples/hello_world/prj.conf'
Configuration saved to '/Users/ochipara/Working/embedded/zephyrproject/zephyr/samples/hello_world/build/zephyr/.config'
CMake Error at /Users/ochipara/Working/embedded/zephyrproject/zephyr/cmake/extensions.cmake:1346 (message):
  No such file or directory: LIBGCC_FILE_NAME: 'libgcc.a'
Call Stack (most recent call first):
  /Users/ochipara/Working/embedded/zephyrproject/zephyr/cmake/compiler/clang/target.cmake:45 (assert_exists)
  /Users/ochipara/Working/embedded/zephyrproject/zephyr/cmake/target_toolchain.cmake:49 (include)
  /Users/ochipara/Working/embedded/zephyrproject/zephyr/cmake/app/boilerplate.cmake:444 (include)
  CMakeLists.txt:5 (include)

galak, 2019-08-06:
> Is the LLVM/ARM support supposed to be functional? I am getting the following error when I try to crosscompile using llvm:

Its not supported out of the box today.  I think this issue was closed not because it was completed, but just tracking it in #3689.