zephyrproject-rtos/zephyr #10157: Use DTS to generate table of supported HW in board documentation

nashif, 2018-09-21:
Right now we generate the table of supported hardware manually and it is difficult to keep it in sync with the actual hardware support in the tree, for example:

https://docs.zephyrproject.org/latest/boards/arm/frdm_k64f/doc/frdm_k64f.html#supported-features

This table could be generated from DTS information using a sphinx plugin that takes the EDTS data as input.

galak, 2018-09-21:
@erwango fyi, another usage for edts

galak, 2022-09-29:
@gmarull here's the issue for generating docs based on DTS info.

erwango, 2024-10-09:
Continuation of the discussion held in https://github.com/zephyrproject-rtos/zephyr/pull/79160 which is addressing this issue.

In [this comment](https://github.com/zephyrproject-rtos/zephyr/pull/79160#issuecomment-2397490964):

> @erwango I was deliberately cautious about using a term along the lines of "available" -- for all the nodes that are not "okay", finding a matching compatible is IMO not enough to automatically assume support is "available". For example, the author of the devicetree may have added a node to document the configuration of a particular device while they were at it and deep in data sheets, even set the proper compatible, but may not have enabled the node just yet due to e.g. a driver limitation. Does that make sense? As discussed, I still would like to indicate that the device may actually work if enabled, but the "may" really needs to be conveyed.

IMO, DT should be used as the actual reflection of what is supported in drivers. In that view, if a node is available in DT, it should match a working configuration. DT shouldn't be used to document wip. If a node is present, but driver is not functional when enabled, this is a bug to be fixed.
This is a simple rule of thumb and I don't think that preventing developers to put wip work in dt files is a real issue.

kartben, 2025-02-28:
Calling victory on this one! 🎉
Infrastructure is in place to mine the list of supported features for all board targets based on zephyr.dts obtained when building a Hello World, and a Sphinx directive allows to output a fancy table of supported features.
https://docs.zephyrproject.org/latest/boards/bbc/microbit/doc/index.html#supported-features

Take that, 6-year-old issue!

closed 2025-02-28 by kartben