zephyrproject-rtos/zephyr #10158: Have support for probot with Zephyr

galak, 2018-09-21:
Probot (https://probot.github.io/apps/) has a number of interesting abilities and it would be nice if we supported it as part of our infrastructure.  Here are some use for:

1. Auto labeling of PR based on files that are modified in the PR (https://github.com/probot/autolabeler)
2. Mark a PR as from a first time contributor
3. Add the ability to rebase from a comment in a PR

We can look into things like:
* Auto closing of old issues/PR
* having comments auto create issues

Plus any other things we want to help automate review and development flow

nashif, 2020-09-26:
with GH actions available, do we really need that?

nashif, 2020-09-30:
superseded by GH actions, closing

closed 2020-09-30 by nashif