zephyrproject-rtos/zephyr #10164: logger sample fails on both native_posix and qemu_xtensa.

nashif, 2018-09-22:
Getting no output on both, see PR #10163 and the CI results.

aescolar, 2018-09-24:
Regarding the native_posix case:
it was working on: 86442038e94815752ecedf0983cf0a8a64eca04e
it was broken on: 68249ce96622979d000c10c89e5e5192802b931b 
(after merging #9362)

nashif, 2018-09-24:
> @nashif regarding qemu_xtensa target. It has no logger backends (as uart is the only available).

I am not sure I am following this, why does it need a special logger backend? it was logging just fine with old logger to the console. How is this different from qemu_x86 or qemu_nios2? What logger backends do those have?

finikorg, 2018-09-26:
qemu uses own UART so it uses usual UART logging backend, @nordic-krch native_posix backend in only for native_posix board configuration.

nordic-krch, 2018-10-02:
@nashif 
> I am not sure I am following this, why does it need a special logger backend? it was logging just fine with old logger to the console.

It's because new logger does not have (yet?) console backend. It has uart backend. So if target has no uart (or native_posix, or rtt when pr comes) it cannot do logging. We should add console backend but we need to be aware that it is byte API so it will slower than optimized uart backend (current uart backend is polling based, it's poor and needs to be improved as well but at least there is room for improvement).

finikorg, 2018-10-02:
native_posix should be working fine, I tested several samples, all OK.

PR #14027 (2019-03-04): logging: add backend for xtensa simulator
Add backend for the xtensa simulator.

Fixes #10164
commit 5cf49956 (2019-03-04): logging: add backend for xtensa simulator

Add backend for the xtensa simulator.

Fixes #10164
closed 2019-03-04 by nashif