zephyrproject-rtos/zephyr #10165: threads: Need to store thread name for debugging purposes

pfalcon, 2018-09-22:
I don't know how this was overlooked for so long, but if corresponding debugging options are enabled, we should store a thread name, just the same as with a corresponding option enabled (CONFIG_THREAD_STACK_INFO), we store stack size info.

This can be see as a follow up to https://github.com/zephyrproject-rtos/zephyr/pull/10044 - with k_call_stacks_analyze() gone, which itself worked around lack of thread names by dealing only with "well known" threads, now it's not easy (if possible at all) to get even the info it provided out, far less good thread info for all threads.

In particular, patches like https://github.com/zephyrproject-rtos/zephyr/commit/0fb9ea03043551c66e6f0ec14c4ef9a001255d7d rather goof it, there it would print something like:

~~~
HCI [info for thread1 which has nothing to do with HCI]
HCI [info for thread2 which has nothing to do with HCI]
HCI [info for thread3 which has nothing to do with HCI]
...
HCI [finally, some thread related to HCI, but user has no idea that it is, because there's no clear identification for a thread at all]
...
~~~

nashif, 2018-09-23:
there is a wip PR https://github.com/zephyrproject-rtos/zephyr/pull/9394

pfalcon, 2018-09-23:
> there is a wip PR #9394

Ah, cool, thanks for working on this. My motion was to close this ticket then, but maybe you'd want to use it as "Fixed: " reference there, so leaving as is, but feel free to close otherwise.

PR #9394 (2018-09-27): Add a new API for setting thread names, useful for debugging/diagnostic
Added k_thread_name_set() and enable thread name setting when declaring
static threads. This is enabled only when THREAD_MONITOR is used. System
threads get a name by default.

With shell sample, this will result for example with:

```

uart:~$ kernel threads
Threads:
*0x00400300 shell     :   options: 0x0 priority: -2
 0x00412040 logging   :   options: 0x0 priority: -2
 0x00401040 idle      :   options: 0x1 priority: 15
uart:~$ kernel stacks
0x00400300 shell      (real size 2044):	unused 1148	usage 896 / 2044 (43 %)
0x004 [... cut]
closed 2018-09-27 by nashif