zephyrproject-rtos/zephyr #10166: config shell & console shell same time cause fail

lgl88911, 2018-09-23:
Hi,pals:
I find eable the config same time, will fail:
```
CONFIG_SHELL=y
CONFIG_CONSOLE_SHELL=y
CONFIG_NET_SHELL=y
CONFIG_BT_SHELL=y
CONFIG_NET_L2_BT_SHELL=y
```
the log:
```
-- Generating zephyr/include/generated/generated_dts_board.h
-- Cache files will be written to: /home/frank/.cache/zephyr
CMake Error at /home/frank/work/project/zephyr/cmake/extensions.cmake:796 (add_subdirectory):
  The binary directory

    /home/frank/work/project/nrf52_moderate/apps/ipsptest/build/zephyr/subsys/shell

  is already used to build a source directory.  It cannot be used to build
  source directory

    /home/frank/work/project/zephyr/subsys/shell

  Specify a unique binary directory name.
Call Stack (most recent call first):
  /home/frank/work/project/zephyr/subsys/CMakeLists.txt:7 (add_subdirectory_ifdef)

```
if disable CONFIG_SHELL or other config will pass.
The means bt shell cmd and shell history function can't co-exist. This is a issue or sepc？Can support bt shell cmd using history function? Thanks !

jarz-nordic, 2018-10-10:
I digged in a bit more.
New shell has different design compering to the old one. We are in the middle of transition so it may not still not work.
Current concept is that you can enable shell commands via Kconfig for NET, BT and so on. But shell instance itself needs to be created separately.

lgl88911, 2018-10-11:
@jarz-nordic Yes, I tested look like you。 BT_SHELL is old shell style , but CONFIG_SHELL is new shell style. Is there a plan covert BT&Net shell to new shell style?

jarz-nordic, 2018-10-11:
@lgl88911 : yes of course. It is ongoing activit. I hope that in 2 weeks we will be able to do it.
Once PR: #10511 is merged it will be very easy task.

lgl88911, 2018-10-11:
@jarz-nordic  Thanks, I‘ll wait your news

jarz-nordic, 2018-10-11:
@lgl88911 : Please track PR #9825. Once it will be merged you shall be able to use new shell.

jarz-nordic, 2018-10-15:
@lgl88911 : Please check now if it is working as you expect it.

lgl88911, 2018-10-17:
@jarz-nordic Update code & test OK, Thanks a lot！
`uart:~$ 
  bt       clear    gatt     help     history  kernel   l2cap    log
  net      net_bt   resize   shell    ticker
`

closed 2018-10-17 by lgl88911