zephyrproject-rtos/zephyr #10186: GPIO callback disable has no effect due to _gpio_fire_callbacks

pdunaj, 2018-09-24:
Function `_gpio_fire_callbacks` triggers callbacks for pins present in mask passed as an argument.

This infrastructure will not work correctly when application disables callbacks for multiple pins from a currently handled callback.

E.g.
4 pin callbacks are registered (A...D). Let's assume 2 are fired (A and B) at the same time.
Interrupt handler is triggered in driver code. It sets bits corresponding for these two pins in the variable passed as an argument to `_gpio_fire_callbacks` and call the function.
As there are two pins callback will be triggered twice (once for pin A and second time for pin B).
The application gets first call (for pin A). In this call the application callback code disables callbacks for  all pins A...D. Application expects no more callbacks for any of the pins.
As `_gpio_fire_callbacks` is still in the loop, after first callback is done, it will trigger second callback for pin B. This happens even-though this callback was disabled.

pdunaj, 2018-11-30:
We are still hit by this...

pdunaj, 2018-11-30:
Fixing this is not trivial really. We changed the application to have one callback for all pins we use. This should do the trick.
I still believe it should be fixed but we will not be affected anymore. If somebody else is hit by that they can do what we did.

pabigot, 2019-01-15:
Perhaps I'm missing something, but I'm surprised this even appears to work.  It's entirely reasonable to disable future callbacks from within the callback's handler; I depend on this in some situations where I have to detect rising and falling transitions on a signal that uses a level trigger (applies to most sensors where an bus transaction is required to read an observation and clear the interrupt).

However  `_gpio_fire_callbacks` iterates over the callback list using the `SYS_SLIST_FOR_EACH_CONTAINER` wrapper, which explicitly isn't safe for the condition where the node being visited is removed from the list while it's being visited.

That could be addressed by using `SYS_SLIST_FOR_EACH_CONTAINER_SAFE`, but the use case in this issue basically empties the list while visiting the first entry, invalidating every pointer that you might have kept a handle on.

One way to deal with that is to mark within callback record itself whether it's participating in a list, and skip invoking the handler if it's not.  Which won't work if a previous callback unregistered the record but re-registered it on another list: suddenly you're iterating over a completely different sequence.

So my tentative resolution would be twofold: (1) switch to the safe list iterator, and (2) document that within a callback the only GPIO callback structure you're allowed to unregister is the one you're running in.

commit d7549570 (2019-02-05): drivers: gpio: fix mis-use of slist API in callback processing

The iterator over registered callbacks failed to account for the
possibility that the callback would remove itself from the list.  If
this occurred any remaining callbacks would no longer be reachable from
the node.  Switch to the slist iterator that is safe for self-removal.

Note that the slist API remains unsafe for removal of subsequent nodes.
Even with the corrected code removal of the next callback registration
(cached in tmp) will result in it being called anyway, with the
remaining unremoved registrations not being called. [... cut]
closed 2019-02-05 by galak