zephyrproject-rtos/zephyr #10188: net: wrong dst mac address in echo reply

rmstoi, 2018-09-24:
If echo request is sent twice to Zephyr with same IP address but with different MAC address, the second echo reply sent by Zephyr has wrong (previous) MAC address.

[echo-reply-wrong-mac.pcap.gz](https://github.com/zephyrproject-rtos/zephyr/files/2411350/echo-reply-wrong-mac.pcap.gz)

jukkar, 2018-09-25:
The ARP cache is used when sending the reply packet, if the cache contains the old address, then that is used.  Usually the host sends gratuitous ARP if its MAC address changes. Currently we do not handle these gratuitous ARP messages, or expire the addresses in ARP cache. So there is definitely some improvements needed here.

jukkar, 2019-02-21:
@rmstoi Is this still an issue?

rmstoi, 2019-02-25:
@jukkar This is still an issue. I compared this case to Linux and Linux sends echo response with correct (updated) MAC.

rmstoi, 2019-02-26:
pcap from running this test with Linux DUT

[linux-echo-reply-correct-mac.pcapng.gz](https://github.com/zephyrproject-rtos/zephyr/files/2904598/linux-echo-reply-correct-mac.pcapng.gz)

PR #13793 (2019-03-02): net: arp: Update the ARP cache if receiving ARP req
If we receive the ARP request, then check if the ARP cache
contains an entry for this IP address already. If it does,
then update the MAC address in the cache.

Fixes #10188
commit 81e83ba4 (2019-03-02): net: arp: Update the ARP cache if receiving ARP req

If we receive the ARP request, then check if the ARP cache
contains an entry for this IP address already. If it does,
then update the MAC address in the cache.

Fixes #10188
closed 2019-03-02 by nashif