zephyrproject-rtos/zephyr #10190: The new shell can only be compiled on boards with SERIAL but it does not set the dependency in its Kconfig

aescolar, 2018-09-24:
The new shell (CONSOLE_SHELL) can only be compiled on boards with SERIAL but it does not set the dependency in its Kconfig options

Related to #10127

jukkar, 2018-09-28:
As discussed this in #10127, the shell should select CONFIG_SERIAL automatically so that it is not needed to be placed in application configuration file.

aescolar, 2018-09-28:
CC @ulfalizer

jarz-nordic, 2018-10-04:
@aescolar @jukkar 
Shell cannot simply select CONFIG_SERIAL. In future when we will have more backends I can imagine that one wants to have only shell on RTT or BT. Hardcoded SERIAL selection makes limited sense in such case.

I will add an option in shell's Kconfig like ACTIVE_BACKEND, when user will have possiblity to select SERIAL or any other...

PR #10379 (2018-10-05): shell: Enable backends via Kconfig file
Added functionality to enable active shell backends via Kconfig
file. When there will be more backends implemented user will
have an option to select only required ones.

It is no longer needed to select SERIAL in prj.conf.

Fixes #10190
commit bc6da1c3 (2018-10-05): shell: Enable backends via Kconfig file

Added functionality to enable active shell backends via Kconfig
file. When there will be more backends implemented user will
have an option to select only required ones.

It is no longer needed to select SERIAL in prj.conf.

Fixes #10190
closed 2018-10-05 by carlescufi